greatscottgadgets/hackrf #260: about libopencm3

bg8wj, 2016-05-17:
when I make after git submodule init and update well, it say 

GENHDR  include/libopencm3/efm32/efm32g/irq.yaml
Traceback (most recent call last):
  File "./scripts/irq2nvic_h", line 174, in <module> main()
  File "./scripts/irq2nvic_h", line 171, in main convert(open(infile), open(nvic_h, 'w'), open(vector_nvic_c, 'w'), open(cmsis, 'w'))
IOError: [Errno 13] Permission denied: './include/libopencm3/efm32/efm32g/nvic.h'
Makefile:58: recipe for target 'include/libopencm3/efm32/efm32g/irq.yaml.genhdr' failed
make: *** [include/libopencm3/efm32/efm32g/irq.yaml.genhdr] Error 1
who can tell me how to solave?
thank you

dominicgs, 2016-05-17:
The permission issue suggests that you may have previously run `make` as root or another user.  Could you check the output of r`ls -l include/libopencm3/efm32/efm32g/nvic.h` to see if this is the case?

dominicgs, 2016-07-07:
@bg8wj Were you able to check the permissions?  Or did you manage to resolve this issue in another way?

dominicgs, 2016-07-13:
Closing this issue as I assume that the problem was resolved.  Please reopen if it was not resolved and you are able to supply the additional information.

closed 2016-07-13 by dominicgs