greatscottgadgets/hackrf #263: firmware upgrade fail

yo2ldk, 2016-06-07:
Hi,

I receive that:  `hackrf_cpld_write() failed: HACKRF_ERROR_LIBUSB (-1000)`
 install libusb-1.0-0-dev is ok..
  please how to finalize the upgrade of firmware?

dominicgs, 2016-06-09:
Which version of firmware are you writing?  How old is your HackRF?  The CPLD code hasn't been updated in the past two years, so this error is safe to ignore if you are updating a relatively new HackRF One.

yo2ldk, 2016-06-12:
Thank you, dominicgs for reply,

  my hackrf is not so old, just taken, I don't know it about CPLD code, now I have on it the new 2015 firmware.
 I try it to do this upgrade, because on WX satellite reception  (137.500 Mhz) I have no signal at all, even now, after upgrade; is some "holes" / gap  on that band?
 also have anyone succeed to TX ?  I take that board for that, I know it is half duplex, that meens now  I have RX, and when push the PTT "button" I have TX; I read somewhere, for that need to power off the board, then we get TX...  how true is that?

dominicgs, 2016-06-14:
There should be no gaps in coverage between 1MHz and 6GHz.  Could you describe the problem in more detail?  Screenshots from gqrx or SDR# would be be helpful.

There is no need to power off the board to switch between TX and RX.

dominicgs, 2016-07-07:
@yo2ldk are you able to supply any more information on this issue?  With the current information I can''t diagnose the problem.

dominicgs, 2016-07-13:
I'm closing this issue, assuming that the problem was resolved. Please reopen if it was not resolved and you are able to supply the additional information.

closed 2016-07-13 by dominicgs