greatscottgadgets/hackrf #269: Frequency gap ?

yo2ldk, 2016-07-18:
I know that HackRF have not so god RF amplifier but, when I try to receive NOAA weather satellites in 137Mhz, I see nothing on waterfall, not even a bit ! 
 With the same equipment , in the same time, but with R820T2  the signal is full.
 So I have  a gap frequency or it is so "deft"  ? or is my unit damage?
Another thing, in the ExtIO dll file can be added something to remove DC ?
 Thank you in advance !

dominicgs, 2016-07-18:
As I asked on your previous issue (#263), could you please provide some screenshots or other information about where you see gaps in the frequency range?  And what you mean by gaps?

HackRF does not have gaps in the frequency coverage, so this could be a software or other issue.  Do you see gaps elsewhere in the range?  Can you successfully receive broadcast FM radio?

What antenna do you have connected?
Which software are you using?
What does `hackrf_info` report?

yo2ldk, 2016-07-18:
this is in windows with sdr# or hdsdr.
FM and 2m band is ok in receiving , just there I have no reception.
I will made some screenshots if is need it
 I not used hackrf_info. i will try

dominicgs, 2016-07-18:
Yes please, could you attach screenshots.

Which antenna is connected?  Anything else?  Filters?  Amplifier?

yo2ldk, 2016-07-18:
antenna is deedicated for 137Mhz; no filters and no LNA this time, just units
[image]
[image]
[image]
[image]

yo2ldk, 2016-07-18:
satellites just passed, but it is some identifiers signals as you see on airspy reception

N8AUM, 2016-07-28:
Alexander, do you have a signal generator ? Also, turn the pre-amp ON. This is what a -130dbm CW signal looks like on mine:

[image]

yo2ldk, 2016-07-28:
Hi Vidas,

  Thank you for reply,
My sig. gen. is under construction now I made one with si5351
I try also with amp (LNA) activated almoust the same , more background noise.
after my vacation  I made an external LNA to see if is better, it seems hackrf is deaf like an old man..

  my best 73 !

rafael2k, 2018-08-24:
Hi all, I bought a HackRF and I have exact the same problem, completely deaf at 200MHz...

straithe, 2021-01-26:
Are you still having trouble? We will close this issue if we don't get a reply within the next month.

closed 2021-02-23 by straithe