greatscottgadgets/hackrf #276: The hackrf one can not been recognized in the Raspberry Pi3

ustbzm, 2016-08-24:
The hackrf one board is plugged in or not, there is nothing in the system message. I use dmesg to print all the system message. Pluging the hackrf one board in USB port, there is no effect on the system message. And use the "lsusb" also print nothing.

schneider42, 2016-08-24:
Some essential troubleshooting questions:
- Which Linux distribution are you running?
- Which modules are loaded?
- Can you post the complete `dmesg` output?
- Do all power LEDs come on as expected?
- Does the USB cable work with another device?
- Does the HackRF work on a PC?
- Does another (USB 2.0) device work on the Raspberry?

ustbzm, 2016-08-25:
I use the Raspbian OS on the Raspberry Pi3. Raspbian is the Foundation’s official supported Operating System.
The complete dmesg output: http://paste.debian.net/791065/
All power LEDs are expected. But the USB LED is off state.
Other USB device, eg mouse keyboard and so on, all work normally.
The hackrf board is OK on a PC.
The command lsusb can not find the hackrf ID.

dominicgs, 2016-08-25:
Do you have access to a powered hub?  I would be interested to see if a powered hub between the Pi and HackRF solves the problem as power is the only issue that I can think would cause this.

davidhubbard, 2016-10-06:
The Raspberry Pi has some very specific power supply requirements, [here](https://www.raspberrypi.org/help/faqs/#power), and if your power supply does not supply clean power (many have this problem) then it will not provide enough power to the HackRF One.

A typiacl solution is to use a powered hub, as @dominicgs suggested.

glabmoris, 2016-12-15:
It works "fine" here with raspbian when using a 2A 5V power supply. I say "fine" because processing power is indeed limited to less than ~5MSPS. Kernel version might be useful to dig deeper into that issue.

Toutoupou, 2017-04-13:
i try to work my hack rf with my raspberry PI. I want to use my hack RF like a receiver FM but my signal is too low (-80/90dB). The problem is from the USB interface ? I need a HUB ?

Thanks for any help!

closed 2019-11-27 by miek