greatscottgadgets/hackrf #278: Binaries are missing in the "latest release"

dmitryelj, 2016-09-09:
Please add a Windows binaries to the "latest release" section.
I cannot find hackrf_transfer.exe anywhere, and it's not comfortable to install half megabyte of IDE tools to build it.

Thanks.

dominicgs, 2016-09-12:
We don't ship binaries as part of our releases, but you can get precompiled libhackrf as part of SDR# or PothosSDR here: http://downloads.myriadrf.org/builds/PothosSDR/?C=M;O=D

dmitryelj, 2016-09-14:
What is the reason that binaries are not available? Copyright issues?

I was needed hackrf_transfer.exe as utility I can use, not libhackrf as a
library I can link :)

James-Brown, 2016-09-14:
Dmitrii – Email me and I’ll send you What I have.
Jim@SETI.Net

From: Dmitrii Eliuseev 
Sent: Wednesday, September 14, 2016 08:34 AM
To: mossmann/hackrf 
Subject: Re: [mossmann/hackrf] Binaries are missing in the "latest release" (#278)

What is the reason that binaries are not available? Copyright issues?

I was needed hackrf_transfer.exe as utility I can use, not libhackrf as a
library I can link :)

closed 2017-01-27 by mossmann