greatscottgadgets/hackrf #281: memory lack

oravard, 2016-09-18:
Hi,

making a infinite  loop with doing only:
 - hackrf_init()
- hackrf_device_list()
- hackrf_device_list_open()
- hackrf_close()
- hackrf_device_list_free()
- hackrf_exit()

shows a memory growing without limit. Is someone experience this issue ?
Is this an issue or do I miss something ?

thanks

Olivier

AlexsJones, 2016-09-19:
```
 Devices are reference counted with libusb_ref_device() and
 * libusb_unref_device(), and are freed when the reference count reaches 0.
 * New devices presented by libusb_get_device_list() have a reference count of
 * 1, and libusb_free_device_list() can optionally decrease the reference count
 * on all devices in the list. libusb_open() adds another reference which is
 * later destroyed by libusb_close().
```

I spent a few minutes looking at this. Suspecting perhaps the usb list wasn't being correctly passed back to lib usb for reference decrement. Having looked at the current lib usb code and the way its used in hackrf I cannot see an immediate issue. 

Can you post your valgrind or gcd output to help narrow down the issue? How much memory are we talking about here, whats the block size?

oravard, 2016-09-20:
I should run a specific test in order to give you the amount of memory which is lost but running the test during one minute takes several 10 Mo.
I take another strategy: doing only hackrf_init, hackrf_device_list, hackrf_device_list_open in the beginning of my program and hackrf_close, hackrf_device_list_free, hackrf_exit at the end. Then a loop doing only hackrf_start_rx (and configured at different frequencies) works without memory lack.
However, I have to modify the hackrf.c : add "do_exit=false" in the hackrf_start_rx function.

regards,

mossmann, 2017-01-27:
Was this perhaps fixed by #246?

dominicgs, 2017-02-03:
It's possible this was fixed in commit a2857f7.  Repeated calls to `hackrf_device_list_open()` would leak memory, but shouldn't now.

miek, 2019-11-25:
I'm going to assume this was fixed in the commit above. If you're still having this issue feel free to re-open or open a new issue.

closed 2019-11-25 by miek