greatscottgadgets/hackrf #284: HackRF and WIN 7 32bit

Rolf9999, 2016-09-23:
Hi,

got the HackRf as a gift. Unfortunately I'm not a Linux guy. How can I instali the HackRF with WIN7 32bit ? The recommendations I found so far are more than 2 years old and do not work on my Laptop. Many thanks for any help.

Rolf

yo2ldk, 2016-09-23:
you can install SDR# from airspy dot com, and there you have hackrf driver as device

dominicgs, 2016-09-23:
I recommend either [SDR#](http://airspy.com) as @yo2ldk recommended, or [HDSDR](http://www.hdsdr.de) if you're more interested in the amateur radio bands.

Rolf9999, 2016-09-23:
Yo2ldk thank you - tried it again, found HacKRF under source menue, select it and try to start, error message = "No device selected"

dominicgs - thank you too, will try HDSDR also

driver is libusb-win32 devices in system setup, there is the HackRF, driver manuf. OpenMoko, Inc, vers. 02.08.2012 1.2.6.0 and status is active, so it seems to be ok

Rolf9999, 2016-09-23:
So found the problem. Was the wrong driver. Problem is solved. Thanks again for your assistances.

Dygear, 2016-09-24:
If you don't like it, I'd be happy to take it off your hands for a fair price.

closed 2017-01-27 by mossmann