greatscottgadgets/hackrf #285: No HackRF boards found - No DFU capable USB device available

thebeanieman, 2016-09-25:
Hello,

I have tried different USB cables but keep getting this message, my Mac computer cant find the HackRF One:

Davids-MacBook-Pro-2:~ David$ sudo hackrf_info
No HackRF boards found.
Davids-MacBook-Pro-2:~ David$ dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb_ram.dfu
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available
Davids-MacBook-Pro-2:~ David$ 

Any tips please?

thebeanieman, 2016-09-25:
Is was a dodgy cable, sorry to take up your time!

closed 2016-09-25 by thebeanieman

buckveez, 2021-11-30:
Thank you!  I had a bad cable as well.

wlopez-enkoding, 2022-03-01:
Bad cable as well :)