greatscottgadgets/hackrf #287: Odd Result from hackrf_set_freq

James-Brown, 2016-09-28:
I get this back from hackrf_transfer -w -n4000 -f 6666666
========================
Receive wav file: HackRF_20160928_083803Z_6666kHz_IQ.wav
call hackrf_sample_rate_set(10000000 Hz/10.000 MHz)
call hackrf_baseband_filter_bandwidth_set(9000000 Hz/9.000 MHz)
call hackrf_set_freq(6666666 Hz/38481896800154409000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.000 MHz)
samples_to_xfer 4000/0Mio
Stop with Ctrl-C

Exiting... hackrf_is_streaming() result: HACKRF_ERROR_STREAMING_EXIT_CALLED (-1004)
Total time: 0.00500 s
hackrf_close() done
hackrf_exit() done
exit
=======================
It only seems to happen on this one frequency. Is this normal?

dominicgs, 2016-09-28:
This certainly isn't normal, I can't seem to recreate the bug with either the last release or the latest code in the git repository.

Could you give me some more details?  Specifically:
Which version of the HackRF tools are you using? 2015.07 ?
Which host operating system are you using?  If Linux, could you tell me the distribution?
Hardware?  PC?  RaspberryPi?

James-Brown, 2016-09-29:
I have no way of knowing the release date of hackrf_transfer.exe that I'm using. I don't recall where I picked up the exe either ( I can't build the tools with my setup).

This might be an 'easter egg' put in by a developer to help debug somewhere along the line. I use 666 for that a lot because it's the mark of the devil <grin> and easy to remember.  It only seems to work with -f 6666666 not a digit more or less or an integer more or less. It's also not sensitive to the value of the other parameters in the string.

This is the full cmd string I'm using:
hackrf_transfer -w -f 6666666 -a 0 -l 0 -i 0 -n 1000 -s 10000000 -b 10000000

The size of the exe is 18,432 bytes but I have no other ID on it. I'm running Windows 10 64 bit.
This is hackrf_info.exe output:
===================
Found HackRF board.
Board ID Number: 2 (HackRF One)
Firmware Version: git-44df9d1
Part ID Number: 0x00674744 0x00674744
Serial Number: 0x00000000 0x00000000 0x457863c8 0x2337151f

closed 2016-09-29 by James-Brown

dominicgs, 2016-09-29:
Hi James,

There's no debug like that in the code but I don't know who built the Windows executable that you have, but I would be surprised if they added in their own debug code.  I'll ask around and see if anyone else sees this issue.