greatscottgadgets/hackrf #290: hackrf_transfer not transmitting/receiving anything

jc1122, 2016-10-22:
I have a Pentoo 2015.0 RC4.6 Release distribution to run hackrf. I updated the firmware to version 2015.07.2 (did only step 1!) according to the instructions here : https://github.com/mossmann/hackrf/wiki/Updating-Firmware . Then I built host software using instructions: https://github.com/mossmann/hackrf/tree/master/host . Then I updated CPLD (step 2 of firmware update). Both firmware and CPLD update seems to be OK. hackrf_info shows an updated device.

I tested the device using GNU Radio Companion to capture and transmit FM radio signal. The results are OK - I could hear the signal from hackrf on a small FM radio device.

I tried to reproduce the capture and transfer with hackrf_transfer:

`hackrf_transfer -r test -f 92000000`

then:

`hackrf_transfer -t test`

I also tried to transmit the file captured with GNU Radio Companion.
Both attempts are successful, as either RX or TX LED is on, however I do not receive any signal on my FM radio device. 
I also tried to play GPS signal with GNU Radio Companion, the attempt was successful - mobile phone could see the signal, however there was no effect when using hackrf_transfer.

What am I doing wrong?

dominicgs, 2016-10-23:
You may want to set some gain settings on both receive and transmit.

It would also be worth taking a look at the received `test` file to see whether or not the data is good.

jc1122, 2016-10-23:
The FM signal is strong, I could capture it with rf, if and bb gain set to zero.
[image]

The spectrum of replayed radio signal (captured with GNU Radio Companion at 96.2 MHz) is attached below:
[image]

[image]

I tried to experiment with hackrf_transfer gain settings, however the result is the same every time. See screenshots below:

[image]

[image]

What else should I check?

jc1122, 2016-10-26:
converting the hackrf_transfer -r output to f32 solved the problem.

closed 2016-10-26 by jc1122

BigNerd95, 2018-01-23:
How did you convert it?

jc1122, 2018-01-23:
use sox, see here:
https://greatscottgadgets.com/sdr/11/