greatscottgadgets/hackrf #291: HACK RFF error in ubuntu 14.04 32 bit

sagniknitr, 2016-11-03:
Please anyone identify and solve this error
[image]
[image]

dominicgs, 2016-11-03:
Try running `sudo ldconfig` to make sure that the system knows where libhackrf is installed.

sagniknitr, 2016-11-04:
Already did that....still the error persists. Is this error because my ubuntu is 32 bit?
I tried the same instructions in ubuntu 64 bit 14.04 and it was working

dominicgs, 2017-01-25:
This looks like different versions of the host tools and libhackrf are installed.  Did you build them both from source?  IS there a distribution provided package installed?

dominicgs, 2017-09-21:
Closing as I believe that I was right about mismatched library and tools.  Also, Ubuntu 14.04 is now old enough that I don't plan to continue supporting it.

closed 2017-09-21 by dominicgs