greatscottgadgets/hackrf #292: FATAL: No supported devices found to pick from

chakpongchung, 2016-11-16:
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.git-28-gc66cb1ba

gr-osmosdr v0.1.4-75-gae686c46 (0.1.5git) gnuradio 3.7.10.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy redpitaya 

FATAL: No supported devices found to pick from.

How we can solve this?

I am running #sudo gnuradio-companion fm_radio_rx.grc

paulathomas, 2016-12-27:
I am having the same problem with a HackRF One despite my set-up working with NooElec SMART. My set up is on Xubuntu 16.10.

dominicgs, 2017-01-30:
Do you have permission to access the HackRF?  Try running `hackrf_info` and pasting the output here.

dominicgs, 2017-09-21:
@chakpongchung, @paulathomas any feedback on this?  I'd like to close this issue.

closed 2017-11-17 by dominicgs