greatscottgadgets/hackrf #293: FATAL: hackrf_set_sample_rate(8e+06) has failed (-1000) HACKRF_ERROR_LIBUSB

chakpongchung, 2016-11-16:
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.git-28-gc66cb1ba

gr-osmosdr v0.1.4-75-gae686c46 (0.1.5git) gnuradio 3.7.10.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy redpitaya 
Number of USB devices: 9
USB device 1d50:6089: 000000000000000014d463dc2f1865e1 match
Using HackRF One with firmware 2015.07.2 

FATAL: hackrf_set_sample_rate(8e+06) has failed (-1000) HACKRF_ERROR_LIBUSB

caoxwei, 2016-11-20:
Do you have any idea of that problem? I come across with the same one,,can anyone help me out?

chakpongchung, 2016-11-20:
So far I tried 3 different ways to install gnu-radio.
 
The environment I have is:
 
cpchung:~$ uname -a
Linux cpchung 4.4.0-31-generic #50-Ubuntu SMP Wed Jul 13 00:07:12 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
 
 
 
 
1.install the one with 16.04LTS distribution
 
2.use PyBOMBS suggested by
 
 https://github.com/mossmann/hackrf/wiki/Getting-Started-with-HackRF-and-GNU-Radio
 
3. Install from source from 
 
http://gnuradio.org/redmine/projects/gnuradio/wiki/InstallingGRFromSource
 
All lead to the same problem:
 
hackrf_set_sample_rate(2e+07) has failed (-1000) HACKRF_ERROR_LIBUSB
 
https://github.com/mossmann/hackrf/issues/163
 
If you read this issue conversation from the maintainer and other users, you will see clearly they admit that it is not working with kernel 3.18 or higher. The reason they claimed is on the kernel driver libusb.

dominicgs, 2016-12-25:
The kernel driver issue was fixed in June 2015 and the 2015.07 release contained the fix.  The output from gr-osmosdr seems to suggest that you are running the up to date version.

Could you attach your HackRF and run `hackrf_info`?

It is possible that power saving is shutting down the HackRF after detection but before you attempt to set the sample rate.  This is still an open issue, but there are work arounds in #243

wdx7266, 2017-04-07:
I have the same problem:

> gqrx 0
> linux; GNU C++ version 4.8.4; Boost_105400; UHD_003.010.001.001-release
> 
> Controlport disabled 
> No user supplied config file. Using "default.conf" 
> gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.10
> built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
> FM demod gain: 1.52789
> IQ DCR alpha: 1.04166e-05
> Using audio backend: auto
> BookmarksFile is /root/.config/gqrx/bookmarks.csv
> getDeviceList : Available input devices: 
>     0 : "HackRF HackRF One 651617" 
>     1 : "RFSPACE SDR-IQ Receiver" 
>     2 : "RFSPACE SDR-IP Receiver" 
>     3 : "RFSPACE NetSDR Receiver" 
>     4 : "RFSPACE Cloud-IQ Receiver" 
>     5 : "RTL-SDR Spectrum Server" 
>     6 : "Red Pitaya Transceiver Server" 
>     7 : "Complex Sampled (IQ) File" 
> Loading configuration from: "default.conf" 
> Configuration file: "/root/.config/gqrx/default.conf" 
> gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.10
> built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya 
> Using HackRF One with firmware 2017.02.1 
> 
> FATAL: hackrf_set_sample_rate(8e+06) has failed (-1000) USB error
> 
> Trying to fill up 1 missing channel(s) with null source(s).
> This is being done to prevent the application from crashing
> due to gnuradio bug #528.
> 
> ******************** 
> ******************** 
> No change in output device:
>   old: 
>   new: 
> 
> Failed to set RX input rate to 8e+06
> Your device may not be working properly.
> 
> IQ DCR samp_rate: 8e+06
> IQ DCR alpha: 1.25e-07
> Changing NB_RX quad rate: 96000 -> 8e+06
> Requested sample rate: 8000000 
> Actual sample rate   : "8000000.000000" 
> New FFT rate: 25 Hz 
> Requested bandwidth: 0 Hz 
> Actual bandwidth   : 0 Hz 
> setFreqCorr : 0 ppm 
> New LNB LO: 0 Hz 
> updateHWFrequencyRange failed fetching new hardware frequency range 
> New mode index: 4 
> FM demod gain: 0.509296
> Filter preset for mode 4 LO: -80000 HI: 80000 
> Genrating taps for new filter   LO:-80000   HI:80000 TW:32000   Taps: 19
> setFftRate to "25 fps" 
> New FFT rate: 25 Hz 
> New FFT rate: 25 Hz 
> setFftSize to "8192" 
> New FFT rate: 25 Hz 

> hackrf_info
> hackrf_info version: git-9bbbbbf
> libhackrf version: git-9bbbbbf (0.5)
> Found HackRF
> Index: 0
> Serial number: 0000000000000000443464c021651617
> Board ID Number: 2 (HackRF One)
> Firmware Version: 2017.02.1 (API:1.02)
> Part ID Number: 0xa000cb3c 0x005e4f58

So, what should I do to slove it?

thank you very much!

dominicgs, 2017-04-07:
Could you try `hackrf_transfer -f 8000000 -r /dev/null`?  I'd like to see if this is a HackRF issue or an gr-osmosdr.

There's still a chance that this is related to power saving.  Have you looked at the workarounds to turn off power saving in #243 ?

wdx7266, 2017-04-08:
Thanks very much! 

1. I try _hackrf_transfer -f 8000000 -r /dev/null_, and I get the following ouput which are same to the problem in [hackrf_set_freq() failed: USB error (-1000) #369].(https://github.com/mossmann/hackrf/issues/369)

>hackrf_transfer -f 8000000 -r /dev/null
> call hackrf_set_sample_rate(10000000 Hz/10.000 MHz)
> call hackrf_set_freq(8000000 Hz/8.000 MHz)
> hackrf_set_freq() failed: USB error (-1000)
> Usage:
> 	-h # this help
> 	[-d serial_number] # Serial number of desired HackRF.
> 	-r <filename> # Receive data into file (use '-' for stdout).
> 	-t <filename> # Transmit data from file (use '-' for stdin).
> 	-w # Receive data into file with WAV header and automatic name.
> 	   # This is for SDR# compatibility and may not work with other software.
> 	[-f freq_hz] # Frequency in Hz [0MHz to 7250MHz].
> 	[-i if_freq_hz] # Intermediate Frequency (IF) in Hz [2150MHz to 2750MHz].
> 	[-o lo_freq_hz] # Front-end Local Oscillator (LO) frequency in Hz [84MHz to 5400MHz].
> 	[-m image_reject] # Image rejection filter selection, 0=bypass, 1=low pass, 2=high pass.
> 	[-a amp_enable] # RX/TX RF amplifier 1=Enable, 0=Disable.
> 	[-p antenna_enable] # Antenna port power, 1=Enable, 0=Disable.
> 	[-l gain_db] # RX LNA (IF) gain, 0-40dB, 8dB steps
> 	[-g gain_db] # RX VGA (baseband) gain, 0-62dB, 2dB steps
> 	[-x gain_db] # TX VGA (IF) gain, 0-47dB, 1dB steps
> 	[-s sample_rate_hz] # Sample rate in Hz (4/8/10/12.5/16/20MHz, default 10MHz).
> 	[-n num_samples] # Number of samples to transfer (default is unlimited).
> 	[-S buf_size] # Enable receive streaming with buffer size buf_size.
> 	[-c amplitude] # CW signal source mode, amplitude 0-127 (DC value to DAC).
> 	[-R] # Repeat TX mode (default is off) 
> 	[-b baseband_filter_bw_hz] # Set baseband filter bandwidth in Hz.
> 	Possible values: 1.75/2.5/3.5/5/5.5/6/7/8/9/10/12/14/15/20/24/28MHz, default <= 0.75 * sample_rate_hz.
> 	[-C ppm] # Set Internal crystal clock error in ppm.
> 	[-H hw_sync_enable] # Synchronise USB transfer using GPIO pins.

2. It is not related to power saving, I think. Because I didn't  install laptop-mode-tools in my Ubuntu 14.04LTS. When I try _apt-get install laptop-mode-tools_, I got the following ouput:

> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> The following packages were automatically installed and are no longer required:
>   kde-l10n-engb kde-l10n-zhcn
> Use 'apt-get autoremove' to remove them.
> The following extra packages will be installed:
>   sdparm
> Suggested packages:
>   hal
> The following NEW packages will be installed:
>   laptop-mode-tools sdparm
> 0 upgraded, 2 newly installed, 0 to remove and 16 not upgraded.
> Need to get 197 kB of archives.
> After this operation, 729 kB of additional disk space will be used.
> Do you want to continue? [Y/n] n
> Abort.

It seems like the  _laptop-mode-tools_ is new for my system? I'm a freshhand, so I'm not sure...

3.Something about my system:

> Operating system:Ubuntu 14.04 LTS

> hackrf_info output:
> hackrf_info version: git-9bbbbbf
> libhackrf version: git-9bbbbbf (0.5)
> Found HackRF
> Index: 0
> Serial number: 0000000000000000443464c021651617
> Board ID Number: 2 (HackRF One)
> Firmware Version: 2017.02.1 (API:1.02)
> Part ID Number: 0xa000cb3c 0x005e4f58

I want to know what should I do to solve this problem?  If you need more information, please tell me.

Thank you very much!

wdx7266, 2017-04-10:
Now, I reweite the Firmware, above situation miss, and I get the following outputs:

> hackrf_transfer -f 8000000 -r /dev/null
> call hackrf_set_sample_rate(10000000 Hz/10.000 MHz)
> call hackrf_set_freq(8000000 Hz/8.000 MHz)
> Stop with Ctrl-C
>  0.0 MiB / 1.000 sec =  0.0 MiB/second
> 
> Couldn't transfer any bytes for one second.
> 
> Exiting... hackrf_is_streaming() result: streaming terminated (-1004)
> Total time: 1.00015 s
> hackrf_stop_rx() failed: USB error (-1000)
> hackrf_close() failed: USB error (-1000)
> hackrf_exit() done
> fclose(fd) done
> exit
>

so, what should I do?

wdx7266, 2017-04-13:
I fix it by replace a USB cable.

It's strange, maybe some line in the cable have some problem, but others are work well. So, I can use a part of function but fail to use others.

closed 2017-11-17 by dominicgs