greatscottgadgets/hackrf #296: again hackrf_open() failed: HACKRF_ERROR_LIBUSB (-1000)

5Volt-Junkie, 2017-01-10:
My HackRF works sporadically. Sometimes, I've to disconnect/connect the USB cable several times until the USB LED lights up. After this, I can work without problems, until reboot of my notebook or disconnect the USB cable. After reboot or disconnect the USB cable, this issue appears again, but sporadically. Sometimes it works at the first time, sometimes I've to plug/unplug the device 20-30 times, until it starts working.

Tried to update the newest Firmware, reinstall the software, resoldered the USB connector on the HackRF, changed USB cable, tried on several notebooks, disabled autosuspend/tlp, active USB-hub, probably tried all tips, I could find on the internet, but same issue. I've also tried it with a HackRF One from my friend, and his HackRF works properly.

When the issue appears, with `hackrf_info` I get the message
```
$Found HackRF board 0:
hackrf_open() failed: HACKRF_ERROR_LIBUSB (-1000)
```

`dmesg` delivers
`can't set config #1, error -110`
or
`can't set config #1, error -71`

I'm using Ubuntu 16.04 LTS

dominicgs, 2017-01-10:
@5Volt-Junkie It certainly sounds like a USB problem.  I usually find it easier to run through the debugging on these via email.  Could you email me (dominicgs at gmail.com) and we'll try to get it taken care of?

dominicgs, 2017-01-25:
@5Volt-Junkie As we have determined that this was a hardware issue and are communicating via email about it, I am going to close this issue.

closed 2017-01-25 by dominicgs