greatscottgadgets/hackrf #297: Very low gain (without RF gain enabled)

xluccianox, 2017-01-13:
Hi, I've recently bought a hackrf one, before I've been playing arround for a while with an R820T2 NESDR Mini Dongle from NooElec.
I'm experiencing an extremely low gain with hackrf compared with a rtl sdr dongle, you can see this clearly in the FM spectrum (I've attached some screenshots), with RF Gain on it looks more alike the rtl sdr dongle output, should use it always on? Cause on Mike's SDR video tutorials he mention not to leave it on (because an strong signal may damage the unit).
The hackrf firmware is up to date.
Could it be a defective unit?

| HackRF with RF Gain = 0 dB  | HackRF with RF Gain = 14dB |
| ------------------------------------- | ------------------------------------- |
| [image]  | [image] |

| NESDR Mini with LNA Gain = 0 dB  | NESDR Mini with LNA Gain = 14db |
| -------------------------------------------- | -------------------------------------------- |
|  [image]  |  [image]  |

Thanks.

PD: Sorry if this issue doesn't belong here, I didn't find another place to ask a solution for it.

candrewneal, 2017-01-13:
I'm having the same problem as you, I'm using SDR# with the ANT500 antenna on Windows 10.

mossmann, 2017-02-01:
What sample rate are you using?  HackRF One's lowest supported sample rate is 2 Msps, and you'll typically get better receive sensitivity at higher sample rates (up to a maximum of 20 Msps).  For best sensitivity, use a high sample rate with a narrow baseband filter bandwidth.

xluccianox, 2017-05-13:
Sorry I didn't answer. I've checked the receiving sensibility and transmission in the RF lab of my university, all 3 hackrfs we tested gave the same values, so there is no problem with mine.
And yes, I was using low sample rates.

closed 2017-05-13 by xluccianox

cyba3r, 2017-08-05:
I have the same issue like xluccianox. What was the solution to fix the problem?

dominicgs, 2017-08-08:
@cyba3r The solution was to use a higher sample rate and a narrower baseband filter.

kouroshdd, 2017-12-10:
Hi i have the same problem with my hack rf its very weak signal but my rtl2832 is very powerfull than hack rf one !!!it is normal or my hack rf have problem please tell me im very nervos

SpiryMedia, 2018-09-14:
HackRF per general has a very weak signal. Extremely weak. Basically deaf. They are selling a very deaf RF.

gpgib6, 2019-03-04:
I can also confirm this. First is from hackrf one, second is from $20 rtl dongle. I will say that using cubicSDR made a big difference though.

[image]
[image]