greatscottgadgets/hackrf #300: Feature Request: use all the 10 bits of the DAC

Paco93, 2017-01-19:
Sorry if this is the wrong place for asking it.
For certain applications 8 bits DAC is quite limiting. As the DACs in MAX5864 are10 bits wide it would be great to have an USB connection mode transferring 16 bits wide samples (accepting a decrease in the maximum usable  sample rate) in addition to the defaul mode transerring 8 bits samples . 
Thanks

mossmann, 2017-01-27:
This is theoretically possible, but it would require a great deal of work.  In my opinion, it would provide very little benefit.  It is already possible to simulate higher resolution at lower sample rates by using a high sample rate with a low bandwidth reconstruction filter (e.g. hackrf_transfer -s 20000000 -b 1750000).

Can you describe a specific example of an application that would benefit greatly from 10 bit TX resolution?

Paco93, 2017-01-27:
My application is in a traffic emulator for random spread spectrum signals  In this application i can generate a large number of cofrequency spread spectrum signals with the number of signals which can change with time to emulate random variabiliti of traffic

Paco93, 2017-01-27:
Sorry i closed inadvertently the message. To conclude I agree that my case is a fringe case. Feel free to close the issue if you think it is not Worth the effort and thanks for the good work you are done.

Radiomix2000, 2017-01-27:
> by using a high sample rate with a low bandwidth reconstruction filter (e.g. hackrf_transfer -s 20000000 -b 1750000
Is this trick possible in gnuradio? 
I also think it's great idea to have 10 bits onboard for more precise accuracy. For some modulations is not necessary to have 8 bit.

mossmann, 2017-01-27:
I'm happy to leave this issue open in case anyone wants to work on it, but I don't plan to spend time on it myself.

mossmann, 2022-09-09:
I'm closing this issue because I no longer believe it to be theoretically possible without a complete reimplementation of our CPLD and SGPIO solutions, removing some other functionality that we rely on. A custom solution for a particular use case may be possible, but it would be external to this repository.

closed 2022-09-09 by mossmann