greatscottgadgets/hackrf #304: operacake -1000 libusb error

f4hbw, 2017-01-24:
Hello everyone, 

I have an error when I try to install my HackRF from official git repository. It works fine on another computer under the same OS Ubuntu 14.04 up-to-date. 

I reinstalled a new version of Ubuntu but can't manage to do it work. 

Does anyone have an idea ? 

Thanks. 

```
~$ hackrf_info 
Found HackRF board 0:
USB descriptor string: 000000000000000014d463dc2f5339e1
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2
Part ID Number: 0xa000cb3c 0x004c4f58
Serial Number: 0x00000000 0x00000000 0x14d463dc 0x2f5339e1
hackrf_get_operacake_boards() failed: HACKRF_ERROR_LIBUSB (-1000)
```

dominicgs, 2017-01-24:
That's my fault, it should fail gracefully. There's nothing wrong with the
hackrf, host library or firmware. However I need to fix the check in the
hackrf_info tool.

dominicgs, 2017-01-24:
You need to update your firmware to match the host code version.  This issue will not appear with the correct firmware.

dominicgs, 2017-01-31:
I have added USB API versioning to better handle this scenario.  Matching firmware and host code versions will still give the best experience, but as of commit 35922fc we handle mismatched versions more gracefully.

closed 2017-01-31 by dominicgs