greatscottgadgets/hackrf #305: Error: HACKRF_ERROR_LIBUSB (-1000)

stackprogramer, 2017-01-24:
hi, i have debian 8, when i connect the hackrf and i type hackrf_info it retruens this as root.
my debian in wmware, i can work it with windows in sdrshrp.
i even tested solution of this link it not fixed...............

> Found HackRF board 0:
Board ID Number: 2 (HackRF One)
Firmware Version: git-7e81647
Part ID Number: 0xa000cb3c 0x00534f4b
Serial Number: 0x00000000 0x00000000 0x406464c8 0x23846d4b
hackrf_get_operacake_boards() failed: HACKRF_ERROR_LIBUSB (-1000)

SonnyRajagopalan, 2017-01-24:
Try the solution in Issue #301

dominicgs, 2017-01-24:
There are two separate issues here:
1) The `hackrf_get_operacake_boards() failed: HACKRF_ERROR_LIBUSB (-1000)` message.  This is a duplicate of issue #304 and I will fix it in the next hour.

2) Unable to access the board as non-root user.  This is a permission issue.  There is a udev rule that gets installed as part of the HackRF host code install process.  It defaults to plugdev group but also supports the usb group.  Check which of these groups exist on your system, then look for `/etc/udev/rules.d/53-hackrf.rules`, check that it references the appropriate group and that you are a member of that group.

stackprogramer, 2017-01-26:
yet i have  `error  hackrf_get_operacake_boards() failed: HACKRF_ERROR_LIBUSB (-1000)`

stackprogramer, 2017-01-26:
finally for not any response i used commit 6b831d31cc15c323ba035183ff425b3980a9378c.
https://github.com/mossmann/hackrf/commit/6b831d31cc15c323ba035183ff425b3980a9378c
now every hackrf_info has good response 
best regards stackprogramer

dominicgs, 2017-01-26:
I'm sorry, I commented on issue #304 but didn't update this one.  As long as your firmware matches the host code version, this error will not occur.

@stackprogramer Did you solve the non-root user access to the HackRF?

stackprogramer, 2017-01-27:
@dominicgs  your welcome, thanks,non-root user access is not solved  yet.but important is for me is working. that works in as root.
best regards stackprogramer

closed 2017-11-17 by dominicgs