greatscottgadgets/hackrf #308: `gqrx` works but not `gnuradio-companion`: `FATAL: Failed to open HackRF device (-1000) HACKRF_ERROR_LIBUSB`

SonnyRajagopalan, 2017-01-25:
Hello,

I have the requisite permissions as my user to run `gqrx` with hackrf (tested it), but when I try to compose a new GRC using `gnuradio-`companion` and execute the flow graph, the execution fails for the following reason:

    Generating: '/mymachine/sdr/gnuradioProjects/flowGraphsFromGRC/top_block.py'

    Executing: /usr/bin/python2 -u /mymachine/sdr/gnuradioProjects/flowGraphsFromGRC/top_block.py

    linux; GNU C++ version 4.8.4; Boost_105400; UHD_3.11.0.git-59-g52ca2e0b

    gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.10
    built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya
    Number of USB devices: 13
    USB device 1d50:0089: 0000000000000000909864c82c3a2fcf match
    Using HackRF One with firmware 2015.07.2
    gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.10
    built-in sink types: uhd hackrf bladerf soapy redpitaya file
    Number of USB devices: 13
    USB device 1d50:0089: 0000000000000000909864c82c3a2fcf match

    FATAL: Failed to open HackRF device (-1000) HACKRF_ERROR_LIBUSB

    Trying to fill up 1 missing channel(s) with null sink(s).
    This is being done to prevent the application from crashing
    due to gnuradio bug #528.

    Traceback (most recent call last):
      File "/mymachine/sdr/gnuradioProjects/flowGraphsFromGRC/top_block.py", line 128, in <module>
        main()
      File "/mymachine/sdr/gnuradioProjects/flowGraphsFromGRC/top_block.py", line 117, in main
        tb.start()
      File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/top_block.py", line 109, in start
        top_block_start_unlocked(self._impl, max_noutput_items)
      File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 4879, in     top_block_start_unlocked
    return _runtime_swig.top_block_start_unlocked(*args, **kwargs)
    RuntimeError: list contains invalid format!

Any insight is appreciated.

Thanks!

SonnyRajagopalan, 2017-01-25:
I don't think the latest version will fix this, I am running v3.7.10.1-182-g59136936. I do not have a permissions issue with LIBUSB as gqrx works just fine with the same user and hackrf board.

dominicgs, 2017-01-25:
How many blocks do you have for the HackRF in the flowgraph?

The log here looks like it has two osmosdr blocks and both are attempting to open the same HackRF device
```
gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.10
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya
Number of USB devices: 13
USB device 1d50:0089: 0000000000000000909864c82c3a2fcf match
Using HackRF One with firmware 2015.07.2
gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.10
built-in sink types: uhd hackrf bladerf soapy redpitaya file
Number of USB devices: 13
USB device 1d50:0089: 0000000000000000909864c82c3a2fcf match

FATAL: Failed to open HackRF device (-1000) HACKRF_ERROR_LIBUSB
```

SonnyRajagopalan, 2017-01-25:
I have one RTL-SDR source and one osmocom Sink (generic source and sink are
not available in my installation done via pybombs). I am stuck on step (9)
of this tutorial:
https://www.csun.edu/~skatz/katzpage/sdr_project/sdr/grc_tutorial1.pdf

dominicgs, 2017-01-26:
That tutorial is using signal sources and GUI sinks, so is unrelated to this issue.

The output from your flowgraph suggests that there are two osmosdr blocks.  Could you share a screenshot of the flowgraph that produces that output?

SonnyRajagopalan, 2017-01-27:
[image]

I have attached the screenshot and the error. It looks like a different error now (I have changed the flow graph to the one in http://www.instructables.com/id/RTL-SDR-FM-radio-receiver-with-GNU-Radio-Companion/?ALLSTEPS).

dominicgs, 2017-01-27:
This isn't giving the same error message as the initial issue report.  That flowgraph was failing to open the HackRF, this one succeeds in opening the HackRF.

If you are interested in a tutorial for getting started with GNU Radio and HackRF, I highly recommend the video series here: http://greatscottgadgets.com/sdr/
One of the first lessosn includes building an FM receiver.

SonnyRajagopalan, 2017-01-30:
I just followed it word for word, the lesson in
http://greatscottgadgets.com/sdr/1/, and my FFT (equivalent to 14:20 in the
video in the aforementioned link) fails for the following reason:

  File "/home/silly/top_block.py", line 100, in <module>
    main()
  File "/home/silly/top_block.py", line 95, in main
    tb.Start(True)
  File
"/usr/local/lib/python2.7/dist-packages/grc_gnuradio/wxgui/top_block_gui.py",
line 74, in Start
    self.start()
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/top_block.py",
line 109, in start
    top_block_start_unlocked(self._impl, max_noutput_items)
  File
"/usr/local/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line
4879, in top_block_start_unlocked
    return _runtime_swig.top_block_start_unlocked(*args, **kwargs)
RuntimeError: list contains invalid format!

My flow graph screenshot is attached:

[image: Inline image 1]

dominicgs, 2017-01-30:
I cannot see an attachment, I believe github strips attachments from email replies to issues, you will need to upload it via the web interface.

SonnyRajagopalan, 2017-01-30:
[image]

Attached is my screenshot. Sorry for the delay!

dominicgs, 2017-01-30:
I haven't seen this error before, but some googling suggests that it is due to mismatched versions of GNU Radio and gr-osmosdr.  This is not a HackRF bug.

I would suggest asking on the GNU Radio mailing list for the solution to this.

SonnyRajagopalan, 2017-01-30:
Thanks, Dominic. I did ask them and they simply called it "operator error."
Let me check again (as I tried a different flow graph last time).

SonnyRajagopalan, 2017-01-30:
Pursuing at http://lists.gnu.org/archive/html/discuss-gnuradio/2017-01/msg00307.html

closed 2017-01-30 by SonnyRajagopalan