greatscottgadgets/hackrf #311: Wait for firmware release

SonyKucing, 2017-01-27:
I wait firmware release because when i compile from git still got error on operacake ..  Thanks....

dominicgs, 2017-01-27:
Did you receive the errors while building the firmware?

Or are you saying you need firmware to match your host code?  If you are compiling from the head of the git master branch, it is expected that you will also build firmware to match the host code.

If you require stable host code and firmware without compiling your own firmware, then you should use the latest release of both.

The operacake warning message only affects hackrf_info and no other functionality.

SonyKucing, 2017-01-27:
yes i got error when building firmware ... this is error message : 
`CMake Deprecation Warning at /usr/share/cmake-3.7/Modules/CMakeForceCompiler.cmake:69 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  toolchain-arm-cortex-m.cmake:27 (CMAKE_FORCE_C_COMPILER)
  /usr/share/cmake-3.7/Modules/CMakeDetermineSystem.cmake:88 (include)
  CMakeLists.txt:26 (project)

CMake Deprecation Warning at /usr/share/cmake-3.7/Modules/CMakeForceCompiler.cmake:83 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  toolchain-arm-cortex-m.cmake:28 (CMAKE_FORCE_CXX_COMPILER)
  /usr/share/cmake-3.7/Modules/CMakeDetermineSystem.cmake:88 (include)
  CMakeLists.txt:26 (project)

-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /usr
CMake Deprecation Warning at /usr/share/cmake-3.7/Modules/CMakeForceCompiler.cmake:69 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  toolchain-arm-cortex-m.cmake:27 (CMAKE_FORCE_C_COMPILER)
  build/CMakeFiles/3.7.1/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:26 (project)

CMake Deprecation Warning at /usr/share/cmake-3.7/Modules/CMakeForceCompiler.cmake:83 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  toolchain-arm-cortex-m.cmake:28 (CMAKE_FORCE_CXX_COMPILER)
  build/CMakeFiles/3.7.1/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:26 (project)
`

dominicgs, 2017-01-27:
These are warnings (which we are looking to clean up) but they do not prevent you from building the firmware.

SonyKucing, 2017-01-27:
yes build firmware success .. but still got error on hackrf_transfer 
hackrf_transfer: symbol lookup error: hackrf_transfer: undefined symbol: hackrf_set_hw_sync_mode

dominicgs, 2017-01-27:
That's a mismatch between your hots tools and your libhackrf.so file.  This can be caused by multiple versions of libhackrf.so being installed.  Do you have a version installed that provided by your distribution's package management software?

SonyKucing, 2017-01-27:
i use kali linux which hackrf and library is installed by default , i compile then use prefix /usr to replace library from my distro

SonyKucing, 2017-01-27:
thanks for your help.. now all is done ...  i has removed hackrf and library from my distro : 
sudo apt-get remove --purge hackrf libhackrf-dev libhackrf0

and hackrf tools running properly ... thanks ....

dominicgs, 2017-01-27:
No worries, I'm glad we could get it running.

I will close this issue now.

closed 2017-01-27 by dominicgs