greatscottgadgets/hackrf #314: hackrf_open() failed: HACKRF_ERROR_LIBUSB (-1000)

ricksantor, 2017-01-29:
First, thank you for taking your time to read this.
When I try to execute commander line like"hackrf_info""hackrf_transfer", it responsed
```
rick@rick:~$ hackrf_info
Found HackRF board 0:
Board ID Number: 2 (HackRF One)
Firmware Version: git-815d1f6
Part ID Number: 0xa000cb3c 0x0059474e
Serial Number: 0x00000000 0x00000000 0x954863c8 0x34309723
hackrf_get_operacake_boards() failed: HACKRF_ERROR_LIBUSB (-1000)
```
or
```
rick@rick:~$ hackrf_transfer -r car.iq -f 27000000 -s 8000000 -n 16000000
call hackrf_set_sample_rate(8000000 Hz/8.000 MHz)
call hackrf_set_hw_sync_mode(0)
hackrf_set_hw_sync_mode() failed: HACKRF_ERROR_LIUSB (-1000)
Usage: blahblahblah

```
The strange thing is that it works good with gqrx and flowgragh in GRC can be run.
I've just install my ubuntu 16.04LTS yesterday and of course things like hackrf,gnuradio,gqrx were installed with it yesterday. Anyone help please.

dominicgs, 2017-01-29:
You're running the latest host code from our git repository, but still the
firmware from the last release. As we add features to HackRF we reach a
point where the two have some incompatibilities.

The solution to this problem is to always use firmware that matches the
version of your host tools. In this case, building the firmware from git.

Going forward I plan to handle mismatched firmware/host versions with a
warning rather than an error.

dominicgs, 2017-01-30:
I'm closing this as it is a duplicate of #304.

closed 2017-01-30 by dominicgs

5frank, 2019-03-18:
`hackrf_info` (might) display this error message if gnuradio-companion is open - sort of expected behavior but I did not realize it. i.e. fixed by closing gnuradio-companion.