greatscottgadgets/hackrf #318: standardize device selection options in hackrf-tools

mossmann, 2017-02-01:
Some tools have -s serial number and -d device index.  Other tools have -d serial number.  Make all tools have the same device selection option(s), perhaps by supporting both serial number and device index with -d.

mossmann, 2017-02-01:
Also check out #252 and verify that device selection is working from gr-osmosdr.  Should there be a function in libhackrf that allows selection by both device index and serial number?

dominicgs, 2017-02-03:
As discussed with @mossmann this is now standardised as -d <serial number> across the HackRF tools.

After the 2017.02.1 release we would like to rework the  way `hackrf_open()` works in libhackrf.  I have opened issue #328 for this.

closed 2017-02-03 by dominicgs