greatscottgadgets/hackrf #321: investigate TX/RX LEDs remaining on

mossmann, 2017-02-01:
Test with both hackrf_transfer and gr-osmosdr and see if there is any improvement that can be made to ensure return to idle transceiver state after TX/RX (apart from improvements that will require rework of transceiver code in firmware already planned for after the 2017.02.1 release).

mossmann, 2017-02-02:
I was only able to induce the fault with kill -9 or with the kill function in GNU Radio Companion (both expected).  Ctrl-C, kill -15, and closing the foreground window in GNU Radio Companion all resulted in the LEDs shutting off.

closed 2017-02-02 by mossmann

kpreid, 2017-02-02:
This can happen easily, it just doesn't in the simple programs GNU Radio Companion can generate. For example, it is sufficient to have Python code containing a reference cycle, which shifts from refcount to general GC mode and thus does not guarantee calling destructors promptly:

```
import sys
from gnuradio import audio, gr, analog, blocks
import osmosdr

class Top(gr.top_block):
    def __init__(self):
        gr.top_block.__init__(self)
        self.source = osmosdr.source('hackrf=0')
        self.reference_cycle = [self]
        self.sink = blocks.null_sink(gr.sizeof_gr_complex)
        self.connect(self.source, self.sink)

top = Top()
top.start()
# then immediately exit because we didn't top.wait()
```

It is of course _possible_ to rewrite the program so that the source block is never referred to by any reference cycle, but this is bad for readability/maintainability (e.g. any interactive program likely has callbacks that set frequency and such).

Firmware noticing when the process stops talking will be a much more robust solution to the problem of "process gone, still in RX/TX mode", but perhaps an at-exit hook of some sort could be a useful interim solution.

It would also be useful to have a way to explicitly terminate RX/TX without destroying the block — such as in a long-running program which wants to manage operating periods in a more straightforward and _guaranteed_ way.

mossmann, 2017-02-02:
Apart from the future firmware change (issue #322), it sounds like your proposed changes are gr-osmosdr things, not HackRF thing.

mossmann, 2017-02-02:
We provide hackrf_stop_rx() and hackrf_stop_tx() in libhackrf.

kpreid, 2017-02-02:
I was primarily intending to address “cannot reproduce”.

Also, gr-osmosdr doesn't have an issue tracker in active use, so there's no actually good place to post such things, and gr-osmosdr is the sole GNU Radio/HackRF bridge, so I figured the notes on what could be done were at least of some interest here.

schneider42, 2017-02-02:
Recently the gr-osmosdr issue tracker moved here: http://osmocom.org/projects/sdr/issues

I think it is reasonable to report this as an issue. If there is some idea on what to do I can spend some time on that.