greatscottgadgets/hackrf #322: firmware transceiver rework

mossmann, 2017-02-01:
Change the firmware to enhance TX and RX functions.  Both full-duplex (with add-on hardware) and half-duplex should be supported in the same firmware.  TX and RX should both avoid the use of stale buffers.  A callback system or other mechanism should allow special processing (e.g. sweep mode or signal source mode) in an extensible way.  The firmware should detect non-responsive USB host applications and return to an idle transceiver state.

dominicgs, 2017-02-03:
Something like this:
[image]

miek, 2021-03-22:
Much of this was merged in #707. #180 can track further work on automatic return to idle.

closed 2021-03-22 by miek