greatscottgadgets/hackrf #324: clock utility

mossmann, 2017-02-01:
There should be a more user friendly way in hackrf-tools to check the status of CLKIN detection and synchronization.  Also implement commands to switch both CLKIN and CLKOUT off and on.  When this is done, change the firmware so that CLKOUT is off by default (to save power and reduce birdies).

This could be a new utility, or we might integrate it into an existing command.  Other things to perhaps consider: support for CLKIN/CLKOUT at frequencies other than 10 MHz, clock calibration, time stamping (with optional GPS), antenna port power control, LED control, GPIO control, hardware sync mode

KC5CQW, 2017-02-23:
Any thoughts, concerns or comments about installing a 50ohm SMA load on the CLKOUT when not used? Would this help with the listed issues of birdies in the mean time? From a hardware perspective, I've been playing with a $15 10MC TCXO from eBay. Would any oscope measurements I could acquire be of any use?
In response to the GPS comments, what about a ublox NEO/LEA-6T incorporated as both a NEMA source and a 10MC clock signal? A software comparator and PLL program could emulate a GPSDO with the GPS mounted on a daughter card and commanded by the HRF.  Internal RFI mitigation may be needed with a GPS mounted on the HRF header blocks(?). Daughter card GPS carrier board could double as a ground plane/emi shield.

KC5CQW, 2017-02-23:
Issues #207 & 216

barto-, 2017-02-23:
@KC5CQW, I played with an ublox LEA-m8T which provides two configurable time pulse outputs. One of the two outputs can be configured to output a square wave with frequencies up to 10 MHz. Unfortunately, the 10 MHz are generated through a fractional division of the main internal clock [(48MHz)](https://www.u-blox.com/sites/default/files/products/documents/Timing_AppNote_%28GPS.G6-X-11007%29.pdf?utm_source=en%2Fimages%2Fdownloads%2FProduct_Docs%2FTiming_AppNote_%28GPS.G6-X-11007%29.pdf). This introduces a lot of jitter making the clock signal useless. Instead, if the clock division is integer (e.g., 8 MHz), the signal is very clean and with a low jitter.
The clock generator IC of HackRF [(si5351c)](https://www.silabs.com/documents/public/data-sheets/Si5351-B.pdf) accepts clocks in the frequency range 10-100MHz. Therefore, something like a PLL-based clock multiplier with a simple firmware modification should do the trick and allow the use of the GNSS-synchronized clock as input clock.

KC5CQW, 2017-02-23:
Sounds legit, so a good VCTCXO around 100Mc phase locked to a 10Mc GNSS with a 10x multiplier and buffer amp would yield a nice 100Mc clock signal. 
The big obstacle is establishing GPS phase lock before the HRF boots. Otherwise, it would sync to the 100Mc vctcxo and then self correct as the GPS signal steers the XO into sync. This would require an already accurate xtal for minimal offset. BTW 96Mc should give a decent multiple of 8Mc with the benefit of lower noise.

miek, 2021-03-22:
`hackrf_clock` is included in release 2021.03.1

closed 2021-03-22 by miek