greatscottgadgets/hackrf #332: hackrf_set_hw_sync_mode() failed

JeroenSlobbe, 2017-02-04:
### Steps to reproduce
When installing the hackRF on a HP laptop (Elite book)  with Ubuntu 16.04.1 LTS I get an error message when running hackrf_transfer. However if I follow the same install procedure on my Desktop (Dell) with Ubuntu 16.04.1 LTS it does work.  (Same HackRF is used)

hackrf_transfer -r test.raw -f 433930000 -x 20

call hackrf_set_sample_rate(10000000 Hz/10.000 MHz)
call hackrf_set_hw_sync_mode(0)
hackrf_set_hw_sync_mode() failed: feature not supported by installed firmware (-1005)

### Expected behaviour
Capturing of the signal

### Actual behaviour
On the laptop I get this error message
call hackrf_set_sample_rate(10000000 Hz/10.000 MHz)
call hackrf_set_hw_sync_mode(0)
hackrf_set_hw_sync_mode() failed: feature not supported by installed firmware (-1005)

### Version information
**Operating system**:

mossmann, 2017-02-04:
What was your installation procedure?
Please provide the output of hackrf_info.

mossmann, 2017-02-04:
By the way, -x does nothing in receive mode.

JeroenSlobbe, 2017-02-04:
installed the libs: git cmake build-essentials libusb-1.0-0-dev g++ libboost-all-dev gnuradio gr-iqbal gr-osmosdr rtl-sdr

git clone https://github.com/mossmann/hackrf.git
cd hackrf/host
mkdir build 
cd build
cmake ../ -DINSTALL_UDEV_RULES=ON
sudo make install

Found HackRF
Index: 0
Board ID Number: 2 (HackRF One)
Firmware Version: 2014.08.1 (API:1.00)
Part ID Number: 0xa000cb3c 0x00434359

JeroenSlobbe, 2017-02-04:
It is weird that it works on the desktop PC but that the laptop fails

mossmann, 2017-02-04:
Your firmware is out of date.  You need to run firmware that matches your host code version.  We provide release packages containing both firmware and host code to make this easy for you.  If you want to run host code from git, you also need to run firmware from the same git commit.

mossmann, 2017-02-04:
What git commit are you running on each host?  I'm guessing they are not the same.

JeroenSlobbe, 2017-02-04:
Fair point,. the desktop was installed about half a year ago,. while I started working on the laptop today,.. where can I find the release that matches the firmware on my machine

mossmann, 2017-02-04:
I recommend running our latest release, 2015.07.2: https://github.com/mossmann/hackrf/releases/download/v2015.07.2/hackrf-2015.07.2.tar.xz

mossmann, 2017-02-04:
Firmware update instructions are here: https://github.com/mossmann/hackrf/wiki/Updating-Firmware

JeroenSlobbe, 2017-02-04:
I share the hackRF with some friends,. so when I update the firmware perhaps I also mess up their systems?  I can simply control the software version to match the current firmware version?

mossmann, 2017-02-04:
Your friends shouldn't have any problems with 2015.07.2 firmware.

JeroenSlobbe, 2017-02-04:
Thanks,. it's working!!

mossmann, 2017-02-04:
Yay! Happy hacking!

closed 2017-02-04 by mossmann