greatscottgadgets/hackrf #339: hackrf_sweep graphing

vsboost, 2017-02-11:
Any tips on graphing the output of hackrf_sweep?
Or can it be piped out to some other application for that purpose.

dominicgs, 2017-02-11:
The CSV output should be compatible with the heatmap script: https://github.com/keenerd/rtl-sdr-misc/blob/master/heatmap/heatmap.py

It can be used from this fork of qspectrumanalyser: https://github.com/mossmann/qspectrumanalyzer

vsboost, 2017-02-11:
Awesome thanks Dom, and thanks for all the hard work getting the new release out.

vsboost, 2017-02-11:
Is is possible to include hackrf_sweep as an option in qtspectrumanalyzer, tried adding it manually but i dont think its passing the correct info to hackrf_sweep.

argument error: '-f 87.0M' invalid parameter(s) (-2)

miek, 2017-02-11:
You should have an option to select the `hackrf_sweep` backend in File -> Settings

vsboost, 2017-02-11:
I only get two options on mine,
rtl_power
rtl_power_fftw

miek, 2017-02-11:
Ah, you need to install the forked version with support for hackrf_sweep: https://github.com/miek/qspectrumanalyzer

vsboost, 2017-02-12:
That would make sense, uninstalled the pip3 one and installed yours but am getting.
Traceback (most recent call last):
  File "/usr/local/bin/qspectrumanalyzer", line 11, in <module>
    load_entry_point('QSpectrumAnalyzer==1.4.0', 'gui_scripts', 'qspectrumanalyzer')()
  File "build/bdist.linux-x86_64/egg/qspectrumanalyzer/__main__.py", line 486, in main
  File "build/bdist.linux-x86_64/egg/qspectrumanalyzer/__main__.py", line 150, in __init__
TypeError: super() takes at least 1 argument (0 given)

dominicgs, 2017-02-12:
QSpectrumAnalyzer only works with python3 (the documentation is somewhat quiet when it comes to that).  I think this error is because of a change between python 2 and 3.

vsboost, 2017-02-12:
Your right, when ran with "python3 setup.py install" it works.
Still not getting anything graphed however, will continue investigating.

vsboost, 2017-02-12:
Got a graph now, just changed the start and stop frequencies and it started.

Thanks again

closed 2017-02-12 by vsboost

vsboost, 2017-02-12:
I know i closed this but just noticed that the gain and ppm settings from qspectrumanalyzer are not being passed to hackrf_sweep

dominicgs, 2017-02-12:
Those are reasonable bugs, but it's probably worth opening an issue against miek's qspectrumanalyzer repo.

I'm not sure that qspectrumanalyzer is the final GUI that I'd want to recommend for this, but it's the one miek got working last summer. I believe there is something else in the works.

vsboost, 2017-02-12:
No probs :-)

hellogeraldblah, 2021-05-08:
Hi! I might be asking this in the wrong thread but did you managed to import the csv file (that the hackrf_sweep outputs) into a program to view it as a waterfall graph? I have managed to output a csv file and generated the heatmap using the heatmap.py from https://github.com/keenerd/rtl-sdr-misc,  but still can't find a program to generate the waterfall diagram.

hellogeraldblah, 2021-05-11:
Thank you for your reply, I have misunderstood certain concepts and have clarified with the documentations online.

lolakhairy, 2023-06-06:
Hi,
I apply to save .CSV file but didn't succeed to plot the row data using python script any help ?