greatscottgadgets/hackrf #341: Capture on external trigger

larkolab, 2017-02-14:
Hello,

I have a candid question, rather than an issue. :)
Is there any way of triggering a capture of IQ samples with the HackRF based on a external trigger, like a GPS PPS?
As there are few pins on the board, I was wondering if this is something that has been thought, or not at all?

The purpose is to be able to precisely measure the arrival time of a signal compared to a reference, like a GPS or something else.

Thanks!

barto-, 2017-02-14:
Hi,

We were able to do exactly what you are asking for. I've never published the code and the results here because I didn't have time for refactoring.
Anyway, we have published a paper in NAVITEC'16 conference, explaining the idea behind our synchronisation technique along with some experimental results. In the end we where able to synchronise multiple HackRFs to a 1PPS signal with a synchronisation error lower than 1 sample.

[M. Bartolucci, J. A. del Peral-Rosado, R. Estatuet-Castillo, J. A. Garcia-Molina, M. Crisci, G. E. Corazza, "Synchronisation of low-cost open source SDRs for navigation applications", Proc. 8th ESA Workshop on Satellite Navigation User Equipment Technologies (NAVITEC), Dec 16 2016](http://spcomnav.uab.es/docs/conferences/Bartolucci_NAVITEC_2016.pdf)

We would be glad to help anyone interested in our technique.

mossmann, 2017-02-15:
There is a new hardware synchronization feature in 2017.02.1 using GPIO that could be adapted to this purpose.  The feature was intended for synchronizing multiple HackRF Ones so that they start RX at the same time.

larkolab, 2017-02-15:
Hi,
Thanks @barto- , I'll read your document!
@mossmann, thanks for the info. Could you elaborate a little about this? What would need to be adapted? So basically I would put a PPS pin on the GPIO input of the hackrf, and the USB capture would be triggered based on this? Can we specify the trigger threshold? I need a precision of 1 microsecond, do you think it is achievable with this feature?

Thanks a lot!

barto-, 2017-02-16:
I didn't have a chance to test this new feature. Has anyone tried this?

mossmann, 2017-02-17:
We're waiting on @dodgymike to write some documentation for the new hardware synchronization feature, but the basic idea is that one HackRF triggers simultaneous initiation of RX on one or more additional HackRFs. Expect that the p.recision would be better than 1 microsecond.  If you want to look at the code, see #302.

dodgymike, 2017-02-17:
I'm getting to it, part-time studying is taking more time than I thought it would! :)

barto-, 2017-02-17:
@dodgymike it seems you are using polling. I tried an approach similar to yours in the past, but I was getting inconsistent results.  Then I discovered it's much easier to do this in the CPLD. The idea is to control the HOST_CAPTURE signal with an external trigger.

dodgymike, 2017-02-17:
@barto- nice paper, did you publish the code anywhere?

barto-, 2017-02-17:
Thanks, I didn't publish the code because it needs some refactoring. I will fork the repo and publish the code, hoping someone could make the code useful for everyone.

barto-, 2017-02-17:
You can find the commit [here](https://github.com/barto-/hackrf/commit/fa6bde951cdd0e2364428a1d628ebee5e94324eb).

The first thing to implement would be a way of enabling/disabling the synchronization via software

dodgymike, 2017-02-20:
Umm, I'm certainly not implementing it - I don't know enough about the CPLD code to do any modifications to the code. Also, synchronisation as it is now works quite reliably for my purposes, and as far as I can tell it should be sufficient for most people's needs.

barto-, 2017-02-23:
Well, ~50 samples of error may be ok or not depending on the application. Anyway, an error below 1 sample would be nice. I'll try to find the time for this...

KC5CQW, 2017-02-23:
Regarding the timing requirements, don't forget the velocity factor of your GPS to antenna via coax AND in circuit delays for triggering the RX via PPS. www.cnsys.com has info about this for their TAC-32 program (Totally Accurate Clock). I use this software with several of my GPS systems and projects.

barto-, 2017-02-23:
Usually, when needed, you can compensate these delays.

jnunezortuno, 2018-01-11:
I'm working on a GPR (Ground Penetrating Radar) application using two HackRF devices. One acting as transmitter and the other as a receiver. They should be syncronized in order to have accurated results. I'm a little bit confused about the multiple-HackRF hardware synchronization feature. 

As far as I read, there are two different strategies:

1. Hardware (CPLD-based) synchronisation of multiple HackRFs #381 
2. Multi device hardware sync #302 

My questions are:

- Are both implemented in the same code? How do you choose each one?

- The strategy nr. 2 needs a firmware update or works with the actual one (2017.02.1)?

- Does either strategy work in GNU-Radio when one device is configured as a transmitter and the other as a receiver? I mean, the transmitted and received samples are synchronized and mantain a constant delay between them?

Thanks in advance!

jnunezortuno, 2018-01-12:
@barto

Now I am using #381. I have updated both SPI flash firmware and CPLD and the _hackrf_transfer_ command seems to work. Now other questions: 

1. Is it possible to synchronize transmission and reception executing following command?

hackrf_transfer **-w** rec1.bin -d HackRF1_serial -H 1 | hackrf_transfer **-r** rec0.bin -d HackRF0_serial -H 1

2. As decribed in http://github.com/mossmann/hackrf/wiki/Multiple-device-hardware-level-syncronisation, Osmocom Source can be used for multi-device streaming but, works at hardware (so it needs #381) or at software level?

3. What kind of modifications should be done in gr-osmosdr in order to achieve this TX-RX synchronization? I have never worked with gr-osmosdr sources...

Thanks!

barto-, 2018-01-12:
@jnunezortuno 
1. Yes, it is possible to synchronise transmission and reception using your command. However, bare in mind that the delays of the TX and RX chains are different and you need to take this into account.

2. I believe that post is outdated

3. Basically you need to call the `hackrf_set_hw_sync_mode` function ([libhackrf](https://github.com/mossmann/hackrf/blob/master/host/libhackrf/src/hackrf.c#L1841)), in order to enable the hardware synchronisation. The relevant code of gr-osmosdr can be found [here](https://github.com/osmocom/gr-osmosdr/tree/master/lib/hackrf)

mrroboto1420, 2018-04-10:
@barto- I am currently working on a project that requires coherent sampling with two HackRF Ones for phase interferometry.  We are currently trying to implement the method here(http://github.com/mossmann/hackrf/wiki/Multiple-device-hardware-level-syncronisation)

We are at the step where we need to modify the gr-osmosdr code to get the GRC flow diagram to trigger the sync.  However, this CLPD method seems better-suited for my application.  

Questions:
(1) What is required software-wise to implement the CLPD method?  Does it involving changing the firmware and VHDL as seen in your paper? 
(2) If so, what is the best way to go about making those changes?
(3) To actually change the gr-osmosdr code, do we basically have to make our own source block that is very similar to the curren osmocom source block BUT internally calls libhackrf to start synchronization?

mrroboto1420, 2018-04-11:
I think I have answered my own questions here, I updated the CLPD via the command specified [here](https://github.com/mossmann/hackrf/wiki/Updating-Firmware).  I am having issues with getting the hackrf_transfer test to work I commented on the other more relevant thread, #381.  As for the second point,  I guess we will have to try and implement the hardware sync command in the osmocom block / source code.

closed 2021-01-23 by mossmann