greatscottgadgets/hackrf #345: undefined symbol: hackrf_init_sweep

gigawatts, 2017-02-16:
### Steps to reproduce
1. compiled and updated new firmware on HackRF
2. compiled and installed new hackrf tools
3. tried to run hackrf_sweep

### Expected behaviour
working

### Actual behaviour
$ hackrf_sweep 
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 0 MHz to 6000 MHz
hackrf_sweep: symbol lookup error: hackrf_sweep: undefined symbol: hackrf_init_sweep

### Version information
**Operating system**: Fedora release 25  kernel: 4.8.8-300.fc25.x86_64

**hackrf_info output:**
$ hackrf_info
Found HackRF board 0:
USB descriptor string: 0000000000000000[redacted]
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1
Part ID Number: 0xa000cb3c 0x00664f63
Serial Number: 0x00000000 0x00000000 0x[redacted]

Not sure what I'm missing here. Any hints?

dominicgs, 2017-02-16:
IT looks like you have two copies of the HackRF tools installed on your system.  The output of hackrf_info has changed recently, and that looks like the 2015.07.2 output.

Did you install a previous version from the fedora repositories?

Could you check /usr/ib and /usr/local/lib for libhackrf ?

gigawatts, 2017-02-18:
Yep, that was it. I did indeed have the 2015 rpm installed Thank you!

closed 2017-02-18 by gigawatts