greatscottgadgets/hackrf #349: hackrf_sweep output order

mrbubble62, 2017-02-16:
The output frequencies are out of order e.g. 1005000000-1010000000  comes before 1000000000-1005000000 

```
hackrf_sweep -f995:1015 -l24 -g24  -1  -w500000
Bin width (454545.454545)
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(5.000 MHz)
Sweeping from 995 MHz to 1015 MHz
Stop with Ctrl-C
2017-02-16, 09:41:03, 995000000, 1000000000, 454545.45, 44, -70.24, -71.40, -91.70, -84.74, -65.27, -66.11, -60.68, -53.81, -42.56, -23.56, -12.05
2017-02-16, 09:41:03, 1005000000, 1010000000, 454545.45, 44, -62.28, -65.10, -64.04, -79.54, -66.34, -67.09, -75.82, -70.07, -75.19, -76.47, -69.86
2017-02-16, 09:41:03, 1000000000, 1005000000, 454545.45, 44, -64.53, -76.48, -76.29, -68.74, -65.92, -65.85, -67.79, -69.82, -63.31, -58.55, -58.66
2017-02-16, 09:41:03, 1010000000, 1015000000, 454545.45, 44, -63.86, -70.77, -80.45, -68.41, -63.53, -63.39, -68.53, -77.65, -71.39, -64.81, -66.83
13.1 MiB / 1.001 sec = 13.1 MiB/second

```
```
hackrf_sweep -f995:1025 -l24 -g24  -1

Sweeping from 995 MHz to 1035 MHz
Stop with Ctrl-C
2017-02-16, 09:58:09, 995000000, 1000000000, 1000000.00, 20, -67.52, -52.38, -40.68, -24.21, -11.38
2017-02-16, 09:58:09, 1005000000, 1010000000, 1000000.00, 20, -60.29, -62.72, -57.93, -54.36, -55.19
2017-02-16, 09:58:10, 995000000, 1000000000, 1000000.00, 20, -60.11, -55.01, -40.73, -22.56, -11.23
2017-02-16, 09:58:10, 1005000000, 1010000000, 1000000.00, 20, -62.37, -68.04, -59.14, -57.11, -57.77
2017-02-16, 09:58:10, 1000000000, 1005000000, 1000000.00, 20, -11.76, -23.97, -41.70, -52.30, -55.54
2017-02-16, 09:58:10, 1010000000, 1015000000, 1000000.00, 20, -59.52, -60.56, -64.04, -63.11, -53.09
2017-02-16, 09:58:10, 1015000000, 1020000000, 1000000.00, 20, -73.32, -65.62, -57.61, -56.39, -67.89
2017-02-16, 09:58:10, 1025000000, 1030000000, 1000000.00, 20, -67.33, -74.21, -63.02, -59.26, -62.16
2017-02-16, 09:58:10, 1020000000, 1025000000, 1000000.00, 20, -65.55, -55.33, -55.37, -71.89, -59.96
2017-02-16, 09:58:10, 1030000000, 1035000000, 1000000.00, 20, -60.22, -71.74, -68.99, -75.60, -67.33
13.1 MiB / 1.000 sec = 13.1 MiB/second
```

dominicgs, 2017-02-16:
This is intentional when using interleaved mode (the default).  Interleaved mode performs an FFT on a 20MHz block and throws out the top and bottom 2.5MHz, as well as the middle 5MHz.  Then we retune by 5MHz and repeat the process.  Finally, we retune by 15MHz and restart the process.

This allows us to cut out the edges of the sample that get caught by the bandwidth filter, and to remove the DC spike in the middle.  We find that this gives us much better data at the cost of some speed.

mrbubble62, 2017-02-16:
OK makes sense now, thanks for the explanation.

closed 2017-02-16 by mrbubble62