greatscottgadgets/hackrf #350: Q. Hackrf as a spectrum analyzer

mrbubble62, 2017-02-16:
What reasonable level of performance could be expected from the hackrf hardware, given a stable low noise external 10 MHz reference and a suitable calibration signal for leveling?

- Sensitivity
- Dynamic range
- Amplitude accuracy
- Phase noise
- Spurious signals

Would it be possible to sync a second hackrf in TX to the sweep function to act as a tracking generator?

Giorgiofox, 2017-02-22:
With the new _sweep function this can be also a very fast spectrum analyzer, if is possible doing a sweep in tx should be a very useful tracking

KC5CQW, 2017-02-23:
I am doing this now as a "poor mans" network analyzer. You will need a RF coupler and good coax/hard line. Minicircuits makes a fine coupler for use up to 3GHz. (eBay for about $30)
Additionally, invest in attenuation devices to protect the DUT and the HRF acting as a spec an.

KC5CQW, 2017-02-23:
I can give further advice via email as this is a development and bug tracking system. Not really a forum type venue. 
My callsign at g mail ;-)

miek, 2019-11-27:
As HackRF One is not calibrated test equipment, we don't guarantee any of those specifications - it would be up to you to evaluate it for your particular use-case.

closed 2019-11-27 by miek