greatscottgadgets/hackrf #352: Unable to use hackrf_sweep - mismatched API version

rouxcorp, 2017-02-21:
### Steps to reproduce
1. Update your HackRF with the last firmware (2017.02.1) as described on the Wiki
2. Run hackrf_info -> the firmware version is equal to 1.00
3. Try to run `hackrf_sweep -f1:3000 -w100000 -l32 -g8`

### Expected behaviour
As describe in the code, hackrf_init_sweep need the 1.02 API version.
```c
int hackrf_init_sweep(hackrf_device* device, ...) {
    USB_API_REQUIRED(device, 0x0102)
}
```

The firmware is only reporting a 1.00 one.
```
$ hackrf_info
Firmware Version: 2015.07.2 (API:1.00)
```

### Actual behaviour
```
$ hackrf_sweep -f1:3000 -w100000 -l32 -g8
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 1 MHz to 3001 MHz
hackrf_init_sweep() failed: feature not supported by installed firmware (-1005)
```

### Version information
**Operating system**: Arch Linux up to date (2017-02-21)

**hackrf_info output:**
```
hackrf_info version: 2017.02.1
libhackrf version: 2017.02.1 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000909864c8383b99cf
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2 (API:1.00)
Part ID Number: 0xa000cb3c 0x00724f56
```

Thanks for all,
And by the way, I love your work ! :-)

Antoine

dominicgs, 2017-02-21:
It looks like the firmware hasn't been updated on the HackRF. Did you reset
the device after the upgrade?

What was he command you used to upgrade it?

rouxcorp, 2017-02-21:
Hello,
I did the flashing process two times. First time, I reset the HackRF with the push-button, and the second time by disconnecting it. In both cases, the firmware read by hackrf_info doesn't change ...

Each time I used theses commands (hackrf_spiflash and hackrf_cpldjtag binaries from Arch Linux and firmware binaries from https://github.com/mossmann/hackrf/releases/tag/v2017.02.1):
- `$ hackrf_spiflash -Rw firmware-bin/hackrf_one_usb.bin`
- Reset
- `$ hackrf_cpldjtag -x firmware/cpld/sgpio_if/default.xsvf`
- The 3 LEDs blinked as expected
- Reset

Did I missed something ?

Giorgiofox, 2017-02-22:
Have you tried in DFU mode?
put your hackrf in DFU mode and:
dfu-util --device 1fc9:000c --download hackrf_one_usb_ram.dfu --reset

dominicgs, 2017-02-22:
For some reason it looks like the firmware isn't being written to flash correctly.  Could you try the DFU method that @Giorgiofox suggested and see if you get the 2017.02 firmware loaded in to RAM?  (note: this method will only load in to RAM, so the firmware will not persist across resets).

grayburn, 2017-02-22:
Ran into the same issue as well. Couldn't flash my Hackrf One using hackrf_spiflash. Firmware version would never change and always said "git-44df9d1 (API:1.00). I am using the release download file of hackrf-2017.02.01.tar.xz

Using DFU mode works well for me, its now showing a firmware version of "2017.02.01 (API:1.02)".

On a side note, if I re-run the hackrf_spiflash command, it puts me back to a firmware version of "git-44df9d1 (API:1.00)

rouxcorp, 2017-02-22:
Hello and thanks to both of you !

The HackRF boots correctly in DFU mode:
```
# dfu-util --device 1fc9:000c --download firmware-bin/hackrf_one_usb.dfu --reset
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=========================] 100%        21992 bytes
Download done.
dfu-util: unable to read DFU status after completion
dfu-util: can't detach
Resetting USB to switch back to runtime mode
```
```
$ hackrf_info 
hackrf_info version: 2017.02.1
libhackrf version: 2017.02.1 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000909864c8383b99cf
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00724f56
```

So ... what can I do ?

rouxcorp, 2017-02-25:
Hello,
By the way, I can't read the SPI:
```
$ hackrf_spiflash -vr /tmp/hackrf_one_usb.bin 
Reading 256 bytes from 0x000000.
```
(the command never ends)

Have a good week-end

dominicgs, 2017-02-26:
@rouxcorp and @grayburn Could you email me (dominicgs@gmail.com) with details of where you bought your HackRF One devices and we'll look at getting this fixed for you.

rouxcorp, 2017-02-27:
@dominicgs I send you a mail a few minutes ago.
Just to warn you: my domain is sometimes considered as junk-mail...

dominicgs, 2017-02-27:
@rouxcorp I received it and replied.

As I'm now in contact with both @rouxcorp and @grayburn I will close this issue.

closed 2017-02-27 by dominicgs