greatscottgadgets/hackrf #356: reason of reduction bandwidth in hackrfone in DVBT-TX?

stackprogramer, 2017-03-06:
### Steps to reproduce
1.gr-dvbt
2.hackrf one used for TX...
3.before it tested with USRP   http://yo3iiu.ro/blog/?p=1220

### Expected behaviour
when i want to transmit gr-dvbt  with hackrf one ,when i set sps to 2M,5M,10M
i expect received spectrum  with bandwidth  2M,5M,10M
### Actual behaviour
but bandwidth was reduction for example 2M is reduced  to 1.4 M
so DVBT is not work.
when i set to 1MB it has some data.......not good qulaity
i know in complex radiation should bandwidth=sps
but why in practical is not happened?????

### Version information
**Operating system**:
gnu linux debian8 , in platform gnuradio

**hackrf firmware version**
https://github.com/mossmann/hackrf/releases/tag/v2017.02.1

stackprogramer, 2017-03-06:
i can transmit dvbt with 1M sps  in gnuradio, when i set 2 Mor 3 M can not  work gnuradio dvbt

sagniknitr, 2017-03-08:
I am having the same problems with gr-DTV example dvb_t transmit GRC flowgraph. Is the sample rate that Hack RF one supports too low for such high data transmission?

stackprogramer, 2017-03-29:
finally my problem is solved,it is related power budget, we decided use amplifier and LNA 
the spectrum is not lost,bandwidth is ok.
best regards stackprogramer

closed 2019-11-25 by miek