greatscottgadgets/hackrf #357: Hack RF one and Raspberry Pi 3

sagniknitr, 2017-03-08:
### Steps to reproduce
1.install gnu radio and osmosdr in raspberry pi
2.run osmocom_fft

### Expected behaviour
Gnuradio Wx-GUI flowgraph with FFT output

### Actual behaviour
"OOOOOOOOOOOO" - overflow symbols in command line terminal

### Version information
Raspbian Jessie 

Is the CPU speed of raspberry pi 3 unable to support Hack RF one sample rate ?

```

dominicgs, 2017-03-08:
The limit is generally in the throughput of the rspberry pi's USB controller.

Could you try running `hackrf_transfer -r /dev/null -s 20000000` and reply with the first 10-15 lines of output?

Toutoupou, 2017-04-12:
Hi, i have some problem to work Hack RF with Raspberry PI. When i configure my HACK RF to receiver FM, i can see on my FFT low signal (-90/-100 dB). I run the same flowgraph (GNU RADIO) on my PC and it works (-30/-40dB). So maybe some data lost on USB transfer ?

Toutoupou, 2017-04-13:
when I run `hackrf_transfer -r /dev/null -s 20000000` terminal reply : 

39.8MiB /1s=39.8MiB/s
40.1MiB /1s=40.1MiB/s

closed 2019-11-27 by miek