greatscottgadgets/hackrf #358: hackrf_sweep stops working after few open/close cycles

pavsa, 2017-03-13:
### Steps to reproduce
1. If you open and stop hackrf_sweep many times, it eventually stops working. the hardware itself stops sending data and the needs to be reset with the button or else no data will be ever received. The hackrf_sweep stars properly, but no data is on the output. 

> call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 0 MHz to 6000 MHz
Stop with Ctrl-C
 0.0 MiB / 1.000 sec =  0.0 MiB/second

> Couldn't transfer any bytes for one second.

> Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total time: 1.00046 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit

I suspect this to be a firmware issue.
The sweep function is a basis for spectrum analyzer I have created [hackrf spectrum analyzer](https://github.com/pavsa/hackrf-spectrum-analyzer) and it runs into the same issue on Windows.

repoalpha, 2018-03-01:
I can confirm something similar in general using the gr-osmmosdr drivers on GNU radio using hackRF. I am calling HackRF via a python script looping for raw IQ samples to a file sink 2000 times (or that was the goal). I get the stdio output on Linux showing the gr-osmosdr driver version, built in source types it supports and the 'Using HackRF One with firmware 2018.01.1 on each iteration the radio unit is called. I don't recall any issue with firmware earlier than 2017.02.1 at all. However 2017.02 and the current 2018.01.1 releases it exhibits an issue where eventually  the loop freezes when calling hackRF repetitively. I note if you ctrl-c after a freeze event it loops 1 iteration but the file contains no data at the freeze point. The file size I am generating on each iteration is a .npy of 200128 bytes. Three things are apparent so far:

1. I note sampling rates less than 8Mhz seem to exacerbate the issue I was using 2.4e6 for a sampling rate but it would stop after a few dozen loops anywhere from 40-70 iterations typically. Taking it to 8Mhz I can get around 800 iterations before it stops. Related who knows, its an observation.

2. The file size generated is perhaps too large per iteration, perhaps a buffer issue or an interruption in the flow due to some other process. I am trying to iterate 2000 times. At 200128 bytes per file its stops randomly at some point not completing the task and freezes. At 8192 bytes per file (GNU radio file sink size of 1024) it makes it all the way to 2000 iterations completing the task.

3. I get no errors from python, or from the hackRF unit or from the OS, it just stops getting data from HackRF and halts.

Given the time it takes to complete 2000 loops trying to find a raw output file size it will accept above 8192 bytes is time consuming. Perhaps there is an explanation for this behaviour?

straithe, 2021-03-20:
We've recently done a release which includes some enhances and fixes to `hackrf_sweep`. Can you please try updating to the latest release to see if that fixes your issue?

closed 2021-04-20 by straithe