greatscottgadgets/hackrf #359: Hackrf_info fails on Raspberry pi 3 ubuntu and Raspbian

kevmiyashiro, 2017-03-15:
### Steps to reproduce
1. Verified latest firmware (2017.02.1), CPLD updated on HackRF, and host software updated.
2. Using Ubuntu MATE 16.04 LTS, also tried with Raspbian Jessie.
3. Made sure all dependencies were up to date.
4.  Note: GNU Radio Companion and OSMOCOM are both able to detect HackRF and receive signals from the HackRF.

### Expected behaviour
When running hackrf_info, expect a return that shows port ID, etc.

### Actual behaviour
When I invoke hackrf_info at command line, I get the following return:
$hackrf_info
hackrf_info version: git-9bbbbbf
hackrf_info: symbol lookup error: hackrf_info undefined symbol: hackrf_library_release

### Version information
**Operating system**:
Using Ubuntu MATE 16.04 LTS, also tried with Raspbian Jessie.

**hackrf_info output:**
hackrf_info version: git-9bbbbbf
hackrf_info: symbol lookup error: hackrf_info undefined symbol: hackrf_library_release

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

dominicgs, 2017-03-15:
Your version of libhackrf doesn't match the version of your hackrf tools.  This is possibly because you have one installed from source and the other from a package?

Try running `find /usr -iname "libhackrf.so*"` and pasting the output here?

kevmiyashiro, 2017-03-15:
Thank you, here is the output

/usr/lib/arm-linux-gnueabihf/libhackrf.so.0
/usr/lib/arm-linux-gnueabihf/libhackrf.so.0.4.0
/usr/local/lib/libhackrf.so.0
/usr/local/lib/libhackrf.so
/usr/local/lib/libhackrf.so.0.5.0

dominicgs, 2017-03-15:
While you're seeing the latest version of hackrf_info, it is picking up the version of libhackrf in `/usr/lib/arm-linux-gnueabihf`  It's fairly likely that this version has been installed by a package manager of some sort.  Try running `sudo apt-get remove libhackrf0`

kevmiyashiro, 2017-03-15:
That did it, thank you!

closed 2017-03-15 by kevmiyashiro