greatscottgadgets/hackrf #360: hackrf host 2017.02.1 FreeBSD build problem (fftw3f)

cederom, 2017-03-18:
Hey :-) I maintain FreeBSD port of HackRF.. I found a `fftw3f` build related problem in 2017.02.1 release host.. the library is found but the build fails.. looks like some CMake related problem..? Please advise :-)

### Steps to reproduce
See Output section at the end..

### Expected behaviour
Build ends without `fftw3f` related problems.

### Actual behaviour
```
Scanning dependencies of target hackrf_debug
[ 31%] Building C object hackrf-tools/src/CMakeFiles/hackrf_debug.dir/hackrf_debug.c.o
[ 37%] Linking C executable hackrf_debug
/usr/bin/ld: cannot find -lfftw3f
cc: error: linker command failed with exit code 1 (use -v to see invocation)
*** Error code 1
```

### Version information
**Operating system**:
```
% uname -a
FreeBSD 0xCFF9 11.0-RELEASE-p8 FreeBSD 11.0-RELEASE-p8 #0: Wed Feb 22 06:12:04 UTC 2017     root@amd64-builder.daemonology.net:/usr/obj/usr/src/sys/GENERIC  amd64
```

**hackrf_info output:**
does not build

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.
```
% ls -al /usr/local/lib/libfftw3f*
-rw-r--r--  1 root  wheel  2854504 Feb 25 02:35 /usr/local/lib/libfftw3f.a
lrwxr-xr-x  1 root  wheel       18 Feb 25 02:35 /usr/local/lib/libfftw3f.so -> libfftw3f.so.3.5.5
lrwxr-xr-x  1 root  wheel       18 Feb 25 02:35 /usr/local/lib/libfftw3f.so.3 -> libfftw3f.so.3.5.5
-rwxr-xr-x  1 root  wheel  1519864 Feb 25 02:35 /usr/local/lib/libfftw3f.so.3.5.5
-rw-r--r--  1 root  wheel    41784 Feb 25 02:35 /usr/local/lib/libfftw3f_threads.a
lrwxr-xr-x  1 root  wheel       26 Feb 25 02:35 /usr/local/lib/libfftw3f_threads.so -> libfftw3f_threads.so.3.5.5
lrwxr-xr-x  1 root  wheel       26 Feb 25 02:35 /usr/local/lib/libfftw3f_threads.so.3 -> libfftw3f_threads.so.3.5.5
-rwxr-xr-x  1 root  wheel    25000 Feb 25 02:35 /usr/local/lib/libfftw3f_threads.so.3.5.5

% pkg info fftw3-float
fftw3-float-3.3.5
Name           : fftw3-float
Version        : 3.3.5
Installed on   : Sat Mar 18 23:06:44 2017 CET
Origin         : math/fftw3-float
Architecture   : freebsd:11:x86:64
Prefix         : /usr/local
Categories     : math
Licenses       : GPLv2
Maintainer     : jhale@FreeBSD.org
WWW            : http://www.fftw.org/
Comment        : Fast Discrete Fourier Transform (Single Precision C Routines)
Options        :
	G77_WRAPPERS   : off
	OPENMP         : off
	OPENMPI        : off
	OPTIMIZED_CFLAGS: on
	SIMD           : on
Shared Libs provided:
	libfftw3f.so.3
	libfftw3f_threads.so.3
Annotations    :
	repo_type      : binary
	repository     : FreeBSD
Flat size      : 4.30MiB
Description    :
FFTW is a C subroutine library for computing the Discrete Fourier Transform
(DFT) in one or more dimensions, of both real and complex data, and of
arbitrary input size. We believe that FFTW, which is free software, should
become the FFT library of choice for most applications. Our benchmarks,
performed on a variety of platforms, show that FFTW's performance is
typically superior to that of other publicly available FFT software.
Moreover, FFTW's performance is portable: the program will perform well on
most architectures without modification.

The FFTW package was developed at MIT by Matteo Frigo and Steven G.
Johnson.  Please send email to fftw@theory.lcs.mit.edu so that we can keep
track of users and send you information about new releases.  The latest
version of FFTW, benchmarks, links, and other information can be found at
the FFTW home page.

WWW: http://www.fftw.org/
```

### Output
```
% tar -xxf hackrf-2017.02.1.tar.xz 
% ls
hackrf-2017.02.1	hackrf-2017.02.1.tar.xz
% cd hackrf-2017.02.1
% ls
COPYING		Readme.md	doc		firmware-bin	host
RELEASENOTES	TRADEMARK	firmware	hardware
% cd host/
% mkdir build
% cd build/
% cmake ..
-- The C compiler identification is Clang 3.8.0
-- The CXX compiler identification is Clang 3.8.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/CC
-- Check for working CXX compiler: /usr/bin/CC -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Using unsigned short
-- Check if the system is big endian - little endian
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "1.3.0") 
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.13
-- Found LIBUSB: /usr/lib/libusb.so  
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Checking for module 'fftw3f'
--   Found fftw3f, version 3.3.5
-- Configuring done
-- Generating done
-- Build files have been written to: (..)/hackrf/release/hackrf-2017.02.1/host/build
% make
Scanning dependencies of target hackrf-static
[  6%] Building C object libhackrf/src/CMakeFiles/hackrf-static.dir/hackrf.c.o
[ 12%] Linking C static library libhackrf.a
[ 12%] Built target hackrf-static
Scanning dependencies of target hackrf
[ 18%] Building C object libhackrf/src/CMakeFiles/hackrf.dir/hackrf.c.o
[ 25%] Linking C shared library libhackrf.so
[ 25%] Built target hackrf
Scanning dependencies of target hackrf_debug
[ 31%] Building C object hackrf-tools/src/CMakeFiles/hackrf_debug.dir/hackrf_debug.c.o
[ 37%] Linking C executable hackrf_debug
/usr/bin/ld: cannot find -lfftw3f
cc: error: linker command failed with exit code 1 (use -v to see invocation)
*** Error code 1

 % grep fftw3f *
CMakeCache.txt:FFTW_LDFLAGS:INTERNAL=-L/usr/local/lib;-lfftw3f
CMakeCache.txt:FFTW_LIBRARIES:INTERNAL=fftw3f
CMakeCache.txt:FFTW_STATIC_LDFLAGS:INTERNAL=-L/usr/local/lib;-lfftw3f;-lm
CMakeCache.txt:FFTW_STATIC_LIBRARIES:INTERNAL=fftw3f;m
CMakeCache.txt:FFTW_fftw3f_INCLUDEDIR:INTERNAL=
CMakeCache.txt:FFTW_fftw3f_LIBDIR:INTERNAL=
CMakeCache.txt:FFTW_fftw3f_PREFIX:INTERNAL=
CMakeCache.txt:FFTW_fftw3f_VERSION:INTERNAL=

% uname -a
FreeBSD 0xCFF9 11.0-RELEASE-p8 FreeBSD 11.0-RELEASE-p8 #0: Wed Feb 22 06:12:04 UTC 2017     root@amd64-builder.daemonology.net:/usr/obj/usr/src/sys/GENERIC  amd64
```

HelterSkelter5, 2017-05-20:
Same problem apparent on macOS compiling  from the same version.  Error is as follows:

`Scanning dependencies of target hackrf-static
[  6%] Building C object libhackrf/src/CMakeFiles/hackrf-static.dir/hackrf.c.o
[ 12%] Linking C static library libhackrf.a
[ 12%] Built target hackrf-static
Scanning dependencies of target hackrf
[ 18%] Building C object libhackrf/src/CMakeFiles/hackrf.dir/hackrf.c.o
[ 25%] Linking C shared library libhackrf.dylib
[ 25%] Built target hackrf
Scanning dependencies of target hackrf_debug
[ 31%] Building C object hackrf-tools/src/CMakeFiles/hackrf_debug.dir/hackrf_debug.c.o
[ 37%] Linking C executable hackrf_debug
ld: library not found for -lfftw3f
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [hackrf-tools/src/hackrf_debug] Error 1
make[1]: *** [hackrf-tools/src/CMakeFiles/hackrf_debug.dir/all] Error 2
make: *** [all] Error 2`

dominicgs, 2017-05-22:
It looks like cmake finds the library and headers (which clang uses) but it fails to find it at link time.

Could you run the make command again but this time set verbose? `make verbose=1`

Also, it may be useful to see whether ldconfig knows about libfftw3f `ldconfig -p | grep fftw3`

cederom, 2017-05-23:
Hello and thank you for reply :-) On macOS I get stucked on cmake:

```
iCeDeROM:build cederom$ uname -a
Darwin iCeDeROM.local 16.6.0 Darwin Kernel Version 16.6.0: Fri Apr 14 16:21:16 PDT 2017; root:xnu-3789.60.24~6/RELEASE_X86_64 x86_64

iCeDeROM:hackrf cederom$ git clone https://github.com/mossmann/hackrf.git hackrf.git
Cloning into 'hackrf.git'...
remote: Counting objects: 12660, done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 12660 (delta 5), reused 4 (delta 4), pack-reused 12653
Receiving objects: 100% (12660/12660), 33.97 MiB | 1.29 MiB/s, done.
Resolving deltas: 100% (9169/9169), done.
Checking connectivity... done.
iCeDeROM:hackrf cederom$ cd hackrf.git/
iCeDeROM:hackrf.git cederom$ cd host/
iCeDeROM:host cederom$ mkdir build
iCeDeROM:host cederom$ cd build/

iCeDeROM:build cederom$ cmake ..
-- The C compiler identification is AppleClang 8.0.0.8000042
-- The CXX compiler identification is AppleClang 8.0.0.8000042
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Using unsigned short
-- Check if the system is big endian - little endian
-- Found PkgConfig: /opt/local/bin/pkg-config (found version "0.29.2")
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.21
-- Found LIBUSB: /opt/local/lib/libusb-1.0.dylib
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found FFTW: /opt/local/lib/libfftw3.dylib
-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   hackrf

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /Users/cederom/cederom/work/CeDeROM/lab/radio/hackrf/hackrf.git/host/build

iCeDeROM:build cederom$ git log
commit 9bbbbbfbfb55c60edadebdc274b6a99281f50a21
Merge: c2e775b 8b85326
Author: Michael Ossmann <mike@ossmann.com>
Date:   Tue Feb 28 16:03:15 2017 -0700

    Merge pull request #355 from dominicgs/firmware_cleanup

    Firmware cleanup
```

HelterSkelter5, 2017-05-23:
> Could you run the make command again but this time set verbose? make verbose=1

```
Scanning dependencies of target hackrf-static
[  6%] Building C object libhackrf/src/CMakeFiles/hackrf-static.dir/hackrf.c.o
[ 12%] Linking C static library libhackrf.a
[ 12%] Built target hackrf-static
Scanning dependencies of target hackrf
[ 18%] Building C object libhackrf/src/CMakeFiles/hackrf.dir/hackrf.c.o
[ 25%] Linking C shared library libhackrf.dylib
[ 25%] Built target hackrf
Scanning dependencies of target hackrf_debug
[ 31%] Building C object hackrf-tools/src/CMakeFiles/hackrf_debug.dir/hackrf_debug.c.o
[ 37%] Linking C executable hackrf_debug
ld: library not found for -lfftw3f
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [hackrf-tools/src/hackrf_debug] Error 1
make[1]: *** [hackrf-tools/src/CMakeFiles/hackrf_debug.dir/all] Error 2
make: *** [all] Error 2
```

> Also, it may be useful to see whether ldconfig knows about libfftw3f `ldconfig -p | grep fftw3`

`-bash: ldconfig: command not found`

There must be a macOS alternative to `ldconfig`?

dominicgs, 2017-05-23:
@cederom That cmake warning is something that I'd like to address, but it's not a failure.  Cmake produces build files and you should be able to run `make verbose=1`
Also, I'd like to verify whether this is a clang problem or a platform problem (I suspect it's something we're not handling correctly for these platforms).  Is it feasible for you to try building using gcc on FreeBSD?

docprofsky, 2017-09-07:
I was able to fix this with these changes.
https://github.com/mossmann/hackrf/compare/687c72fc37af42ad2e1cb06a54c8a60a11dc601e...docprofsky:operacake-sctimer#diff-b4f0669c774466f10e5ed3dc789f2fc1

I believe the problem is that `FindFFTW.cmake` finds `fftw3` (`libfftw3.dylib` on macOS) not `fftw3f` (`libfftw3f.dylib`) which is what `hackrf_sweep` needs to be linked against.

Upon additional investigation, changing `fftw3` to `fftw3f` on [line 19 of `FindFFTW.cmake`](https://github.com/mossmann/hackrf/blob/84b29ac5716c6cf30dc5332c93a582413104b31f/host/cmake/modules/FindFFTW.cmake#L19) should fix this. However, this may break the linking of FFTW for unaffected systems.

dominicgs, 2017-09-11:
Yes, this definitely looks like we need to fix the `FindFFTW.cmake` file.  I'll assign this to myself and try to fix it before we next release.

dominicgs, 2018-04-05:
There was a pull request to fix this ( #465) but it broke other builds.  We need to work out how to fix this without breaking other things.

closed 2021-12-03 by straithe

cederom, 2021-12-03:
Thank you guys! :-)

straithe, 2021-12-03:
You're welcome! Hopefully the cmake changes are sufficient! Please open a new issue if anything else needs to be addressed.

cederom, 2021-12-03:
I will verify this after some other builds problems are fixed :-)
* current release 2021.03.01:
```
hackrf-2021.03.1/host/hackrf-tools/src/hackrf_sweep.c:110:24: error: use of undeclared identifier 'SAMPLES_PER_BLOCK'
uint32_t num_samples = SAMPLES_PER_BLOCK;
```
* current head master:
```
CMake Error at CMakeLists.txt:28 (include):
  include could not find requested file:

    set_release

CMake Warning (dev) at CMakeLists.txt:29 (add_definitions):
  Policy CMP0005 is not set: Preprocessor definition values are now escaped
  automatically.  Run "cmake --help-policy CMP0005" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.21)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.
```

I will open separate Issues for the above :-)

Thank you for your time! :-)

straithe, 2021-12-03:
Thank you. We'll continue conversation in the new issues.

cederom, 2021-12-03:
FFT3W IS FIXED! TANK YOU! :-)
Confirmation here: https://github.com/greatscottgadgets/hackrf/issues/1007

PR #664 (2021-12-03): Cleanup of host software CMake build system
Hi all!  Recently we got a HackRF at my lab (USC RPL), and while we're having a really fun time playing around with it, we did have a few issues at the start getting the host software set up -- especially on Windows with MSVC.  It wasn't anything we couldn't deal with, but I did see a few places where the CMake build system could be improved, so I thought I'd try to clean it up a bit.  I incorporated a few bits and pieces from other projects I've worked on (mainly LibraryUtils.cmake, and a much less sophisticated version of FindFFTW), but also wrote  [... cut]