greatscottgadgets/hackrf #361: wrong localtime argument type in hackrf_sweep causes crash on win 10

jaredd, 2017-03-19:
### Steps to reproduce
1.  Build and install from git commit 9bbbbbfbfb55c60edadebdc274b6a99281f50a21 with MSVC 2017 community on Win 10 x64
2. run hackrf_sweep
3. 

### Expected behaviour
Text output corresponding to spectrum captures

### Actual behaviour
No output, utility crashes with bad conversion argument to hackrf_sweep.c:315.

The problem stems from the fact that the definition of timeval (time_stamp type) contains long ints which are sized at 4 bytes.  The default localtime argument in the described configuration is 8 bytes.  The _USE_32BIT_TIME_T option is not allowed with _WIN64 defined.  The current code outputs an invalid fft_time on line 313 which is translated to a garbage string by strftime which triggers the error on 315.  The current workaround is to replace line 313 with the lines:

			time_t time_stamp_seconds = time_stamp.tv_sec;
			fft_time = localtime(&time_stamp_seconds);

where time_t is the correct type (_int64 or something like that) to pass to localtime.

### Version information
Win 10

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

guruofquality, 2017-11-30:
A user also reported this to me: https://groups.google.com/forum/#!topic/pothos-users/uLykVWUI-jw I definitely agree about fixing the type/pointer abuse. Can I do anything to help to speed this along -- like a pull request, or testing a branch with the fix on win10?

dominicgs, 2017-12-03:
Fixed and pushed: https://github.com/mossmann/hackrf/commit/a1f9361f78df45c69cfac725b2249a0fd9f147e1

@guruofquality Are you able to test this on Win10?

dominicgs, 2018-04-01:
Did we determine if this was producing the expected output?  Or at least consistent output across platforms?

closed 2021-03-03 by straithe