greatscottgadgets/hackrf #362: hackrf_sweep output missing bins on high system load

michaellass, 2017-03-26:
The number of frequency bins output by hackrf_sweep reduces when the system is under high load. Most likely, frequency bins are just dropped in this case. This showed to be a problem in qspectrumanalyzer, because here a data structure of fixed size is allocated after the first sweep and the output of hackrf_sweep was expected to be of same size for subsequent sweeps: xmikos/qspectrumanalyzer#26

For qspectrumanalyzer, there is now a workaround in place which drops an entire sweep if the number of bins is not equal to that from the first sweep but this still fails when bins are dropped during the very first sweep.

Is there anything which can be done regarding this issue on the side of hackrf_sweep? Or is there a simple way to determine the number of bins that *should* come from hackrf_sweep, given a set of parameters?

ZeroChaos-, 2017-03-29:
ping @miek

ZeroChaos-, 2018-08-21:
ping @miek 

I have a new computer, a Ferrari compared to my old one, still have this issue.

dominicgs, 2018-09-06:
A reminder to myself, from the discussion in #hackrf 
```
dominicgs> We can take a look at the issue. Is it just high system load that triggers it?
Zero_Chaos> dominicgs: I have a brand new 3000 dollar laptop.  I'm pretty sure it's not a system load issue
dominicgs> Zero_Chaos: So, the issue says it appears under high system load, but I've never been dominicgs> I'm happy to look at it, but I'm going to need some more information about the system at the time that it happens
dominicgs> Is it consistently a problem? Or intermittent?
Zero_Chaos> dominicgs: consistent. hackrf_sweep seems to return a different number of samples all the time, while qspectrumanalyzer expects the same number
Zero_Chaos> dominicgs: set qspectrumanalyzer to go 1-6000 once per second and I get more failures than successes
Zero_Chaos> 599196 bins coming from backend, expected 601200
Zero_Chaos> dominicgs: bin size 10khz fails horribly, 100 seems to work
```

straithe, 2021-03-24:
We've recently done a release which includes some enhances and fixes to hackrf_sweep. Can you please try updating to the latest release to see if that fixes your issue?

michaellass, 2021-04-06:
Sounds good. I have since then changed my system and will need to find some time to test if I can still reproduce the original problem and if the new version fixes it. I'll definitely test this but I may need a couple of weeks.

straithe, 2021-04-06:
Thanks for the update. I'll check back in in a month if you haven't sent another update by then.

michaellass, 2021-05-02:
So, I finally managed to retest this. The hackrf version does not make any difference.

**But**: Revisiting this issue, I think this is actually not bug in hackrf_sweep at all. I rather think that the problem is that qspectrumanalyzer (being implemented in Python) just cannot process the data quick enough.

qspectrumanalyzer connects to the output of hackrf_sweep via pipe (https://github.com/xmikos/qspectrumanalyzer/blob/625ba9d698cc9bbb73efa3dc4d603c9ddc40706b/qspectrumanalyzer/backends/hackrf_sweep.py#L112). As soon as all data from one sweep has been received, the data is processed (in particular, sorted: https://github.com/xmikos/qspectrumanalyzer/blob/master/qspectrumanalyzer/backends/hackrf_sweep.py#L136). If during this time the output of hackrf_sweep continues and fills the buffer of the pipe, this creates backpressure and hackrf_sweep will probably block.

One can kind of reproduce this using `pv` to throttle the pipe:

```
hackrf_sweep -f 100:1000 -w 10000 -l 16 -g 24 -N 3 | wc -l
540

hackrf_sweep -f 100:1000 -w 10000 -l 16 -g 24 -N 3 | pv -L 1M | wc -l
528
```

So, I currently tend to just closing this issue. This would probably have to be fixed in qspectrumanalyzer and there are already workarounds available (throttling & dropping of incomplete sweeps).

straithe, 2021-05-02:
Thank you for the update. I will close this issue as you have suggested.

closed 2021-05-02 by straithe