greatscottgadgets/hackrf #364: sweep for transmit

yhyamak, 2017-03-29:
is there any way to modify sweep.c to make it in tx mode 
is there  a plan to do that or is this possible ?
because i have to hackrf and i want to make one of them a tracking generator  and another one as a spectrum analyzer

f4grx, 2017-06-18:
you can use osmocom_siggen from gr-osmocom, but it needs a gui.
there is also osmocom_siggen_nogui that you can use as a base to make a python/gnuradio based tracking generator.

petem001, 2018-01-05:
yhyamak have you been able to do a sweep generator ?

yhyamak, 2018-01-05:
Yes ...but not as real tracking generator
I use osmocom_siggen but its not good enough
It has a bandwidth about 10 mhz
But i need more than that

petem001, 2018-01-05:
Ok thanks for the feed back, could it be that the bandwith limitation is caused by the max transfert rate is set at 10 mb/s in the python script? will check on this. Thanks again for the feed back

yhyamak, 2018-01-05:
You are welcome .
I think it mist be edited in frimware level 
It is same as RX .it just need to make a new function in sweep.c

feramo, 2018-07-13:
Hi yhyamak, not sure if you are still interested in making a tx sweep, but it is not necessary to make changes at firmware level. I am starting a new sw for doing this sweep, if you are interested I can let you know when it gets ready. Actually, I already have one done, but it is a console program that calls osmocom_siggen, so it is takes a while to start, not suitable for a tracking application... What I am doing now is a mix of sweep and transfer tools.

jinnu92, 2019-02-27:
>hi Feramo sweep transfer is completed or not I am very interested in this.

yhyamak, 2019-03-16:
Hi feramo ,
sorry for long time to reply ,yes I'm very intresting to solve this issue ,
so if you  can let you know when it gets ready i will be grateful to you ,
and i  hope to tell me what you have done until now ,if i can help to develop the firmware  i will do my best

cyberkul, 2019-06-19:
I`m also interested in this. So far it`s just an idea for doing RF-antenna measurements. I want to have the receiver in a drone, and transmitter at the ground,  flying a predermined pattern, doing signal strength measurement at waypoints. If I could do syncronized sweeps on both receiver and tranceiver, I could get antenna pattern diagrams for a lot of frequencies "at once". If this is not possible I could still sweep either the receiver, or transmitter, and step the one that is not sweeping.

luoGanda, 2019-11-28:
i am not sure - but this is probably not a lot of work to write this peace of code.
I would write this peace of firmware - but i don't have knowledge how hrfLib/rfPackets work and currently no time for it.

Some mentioned osmocom_siggen.
Yes, but that is a simulation of what some peace of a code could do.
Currently hardware is capable of something that there is no software for it.
It would be quite handy option - i hope this will be a part of updated firmware.

Also, some code could be checked directly from portapack's code for this - but i don't know if there it's also simulated through already made library calls, or written from scratch for real.