greatscottgadgets/hackrf #365: Antenna to be used with HackRF

Rohan131293, 2017-03-30:
Hello,
In order to cover LTE 800 and 2600 both with same antenna using HackRF ,what is the best choice?Which antenna you have used while working with HackRF in this band.What parameters other than frequency to be considered while deciding antenna?The default antenna with HackRF is ANT500 which supports till 1.1GHz only.Is it totally useless above this limit?
Thanks....

dominicgs, 2017-06-29:
Your choice of antenna depends on a number of things: frequency, pattern, size, etc

If you want to cover a wide range of frequencies, I like the log periodic antennas from [WA5VJB](http://www.wa5vjb.com/index.html), although they are directional.

The ANT500 isn't completely useless above 1.1GHz, but the performance drops with increased frequency as the antenna is not the correct size for those frequencies.

FFY00, 2017-06-30:
I recommend a Taoglas TG.09.0113. It's cheap and really good. I own 2 of those.

https://www.digikey.com/product-detail/en/taoglas-limited/TG.09.0113/931-1038-ND/2332665

You can always buy a generic GSM/LTE antenna from ebay.

closed 2017-11-16 by dominicgs

komalkumar6655, 2019-11-07:
I want to transmit GPS frequencies using Hack Rf .
Can anyone please suggest which type of antenna to buy.
Thanks.