greatscottgadgets/hackrf #366: when update to new firmware >>Part ID Number: 0xa000cb3c 0x00534f4b

stackprogramer, 2017-04-04:
### when i update firmware hackrf_one :
1.i used this link for update firmware  https://github.com/mossmann/hackrf/releases/tag/v2017.02.1
2.product id and vendor id is change?? 	/\00/\   

> > root@debian:/home/admin# hackrf_info
> hackrf_info version: 2017.02.1
> libhackrf version: 2017.02.1 (0.5)
> Found HackRF
> Index: 0
> Serial number: 0000000000000000406464c823846d4b
> Board ID Number: 2 (HackRF One)
> Firmware Version: 2017.02.1 (API:1.02)
> Part ID Number: 0xa000cb3c 0x00534f4b
> 

3.rule usb that i set is not worked.................

### Expected behaviour
when i am not in root user i set rule group that i can acces hackrfone.

### Actual behaviour
but in ordinary user mode it returns 

> admin@debian:~$ hackrf_info
> hackrf_info version: 2017.02.1
> libhackrf version: 2017.02.1 (0.5)
> Found HackRF
> Index: 0
> hackrf_open() failed: Access denied (insufficient permissions) (-1000)
> 

### Version information
debian os > libhackrf version: 2017.02.1 (0.5)

for settig rule i used different modes:
idvendor:idproduct= a000cb3c :00534f4b
idvendor:idproduct= cb3c :00534f4b
idvendor:idproduct= cb3c:534f4b
i too set group to my group user...............
i saw too  this link
http://wiki.openmoko.org/wiki/USB_Product_IDs
**my question why  product /vendor id is changed
why rule in past worked now that i set doesn't work..................**
thanks in advance best regards stackprogramer

stackprogramer, 2017-04-04:
finally i find solution ,i should add OWNER="" to my rules.every thing work like charm.

> ATTR{idVendor}=="1d50", ATTR{idProduct}=="6089", SYMLINK+="hackrf-one-%k", MODE="760",OWNER="admin", GROUP="admin"

thanks 
best regards stackprogramer

korczis, 2017-05-20:
Hi, I am having same troubles on mac. Any ideas how this can be solved?

JamesTheHacker, 2017-07-14:
Same problem. On Mac too.

dominicgs, 2017-07-14:
This issue was a permissions issue related to the udev rule that we install having a different group to the user.  That does not apply to OS X.

When you say you're seeing the same problem, do you mean that you're getting `hackrf_open() failed: Access denied (insufficient permissions) (-1000)` from `hackrf_info` ?

dominicgs, 2017-11-16:
@JamesTheHacker @korczis Did you solve your issue?  Or do you have any more information to help debug it?

johmedina, 2018-07-31:
I have the same issue on ubuntu, I'm getting `hackrf_open() failed: Access denied (insufficient permissions) (-1000).` I installed gnuradio and gr-osmosdr through pybombs so I cant find where my udev rules are. My /etc/udev/rules.d is empty. How do I solve this problem then?

dominicgs, 2018-08-01:
You can find the file here: https://raw.githubusercontent.com/mossmann/hackrf/master/host/libhackrf/53-hackrf.rules  and copy it to `/etc/udev/rules.d/`

SorenAndreasen, 2019-06-26:
i'm having this issue on windows 10. Any ideas how to solve this, please ?

SorenAndreasen, 2019-06-26:
okay, setting "hackrf=0" in device arguments and replugging hackrf one did the trick

closed 2019-11-27 by miek

Dygear, 2019-12-21:
@SorenAndreasen's fix also works on macOS Catalina. Just an FYI for everyone else.

lindes, 2021-02-19:
@SorenAndreasen wrote:

> ... and replugging hackrf one did the trick

For what it's worth, the reset button is a friendlier way to do this.

That said, hitting the reset button after @dominicgs's suggestion of copying in a 53-hackrf.rules file did the trick for me.  Thanks!

kobialka, 2021-05-25:
> ### when i update firmware hackrf_one :
> 
> 1.i used this link for update firmware https://github.com/mossmann/hackrf/releases/tag/v2017.02.1
...
> thanks in advance best regards stackprogramer

Hey, 
outdated or not... what worked for me was to change ._rule_ file name prefix from "stock" 52- to 92- so as I understand it gets done with lower priority, means later in time or maybe after something else. 
At my system it works A->B->A, just by changing the prefix.
BR
MK