greatscottgadgets/hackrf #367: hackrf_open() failed: Resource busy (-1000)

kevmiyashiro, 2017-04-06:
Question: is there a way to send a "soft" (command line) reset to the HackRF?  I'm having intermittent problems with the HackRF locking up, and would prefer to not have to cycle power every time the device stops responding.

### Steps to reproduce
1.  calling hackrf_sweep through a simple "infinite" do while loop in a shell script
while true; do
hackrf_sweep -f 20:100 > out.txt
sleep 2
done

### Expected behaviour
generate / update out.txt for each pass.  Note: each pass is taking approx 1 second, so a 2 second wait before passing through the loop again seems sufficient.

### Actual behaviour
**sometimes**, after going through a number of passes through the loop, the HackRF gets "busy" and the hackrf_open() fails.  Once the device is busy, it remains busy until I kill the loop.  Usually waiting a few seconds releases the HackRF and I can run the loop again, but sometimes I need to do a hard reset and cycle DC power.  

### Version information
Running on Ubuntu 14.04 and Ubuntu MATE (Raspberry pi)

**hackrf_info output:**
none relevant

### Output

dominicgs, 2017-04-06:
Assuming that you are running the latest firmware and host code, you can issue a reset from the host using `hackrf_spiflash -R` (it's part of the spiflash tool because it's most useful for resetting after flashing).

However, if the HackRF isn't responding to USB commands, it will quite possibly not respond to the reset from the host.  I'd be very interested to know if this solves your problem.

Additionally, I'm unsure what you are trying to achieve with the repeated running of hackrf_sweep, by default it should run continuously.  Is hackrf_sweep failing after a short time?

kevmiyashiro, 2017-04-06:
Good to know; unfortunately, as you suspected, once the bus locks out, I can't send any more commands in (tried to execute the spiflash from another terminal, says the resource is busy).  So power cycle it is.

Regarding the infinite loop, it's something we're tinkering with, we needed a way through a script to run the sweep continuously but then asynchronously receive updates to the sweep parameters via a file and then retrigger the sweep.  So as a quick and dirty, setting up the infinite loop got the job done.

In answer to your question about the sweep failing, no, when running it the standard way, it's run for over an hour continuously with no problems.

dominicgs, 2017-04-06:
I haven't found myself in that particular situation, but I believe there are some things that the host OS can do, I found that this worked for me on my box (but my HackRF was receiving, not hung): http://askubuntu.com/a/61165

Good luck.

kevmiyashiro, 2017-04-07:
Interesting, thank you for the lead, will take a look at that.

We're running into a handful of "sometimes this happens" stuff, wanted to hear if you have any thoughts on why it might be happening so we can figure out a mitigation strategy for it.

BTW, I forgot to mention that we're using one-shot mode, embedded in the infinite loop, partially because we need to control the size of the output file.  By doing successive one-shots, we can easily overwrite the file with the last sweep, versus the appending that seems to happen with a continuous hackrf_sweep.

That being said, "sometimes" when we're in infinite loop, one-shot mode, the hackrf will eventually (usually 10's of passes later) reach this point where it's still able to execute the hackrf_open() command, but the stdout reads

0 total sweeps completed, 0 sweeps/second
0 total sweeps completed, 0 sweeps/second
over and over again....

it never seems to break out of this mode until we kill it / ctrl-c it.  The difficulty in diagnosing this is that it only happens "sometimes," but I believe, it will "always" eventually happen.

Any thoughts?

dominicgs, 2017-04-07:
Have you tried adding the reset (using `hackrf_spiflash`) with this?

At first look, I thought this might have been a problem with the transceiver staying in sweep mode after hackrf_sweep exits, but we actively reset the transceiver mode as we exit, or else we print an error, so I think that's unlikely.

kevmiyashiro, 2017-04-07:
The hackrf_sweep is being called in a shell script, and I'm trying to set up logic to trap errors coming out from hackrf_sweep and exit my infinite loop, but it's not working, including trying the _spiflash reset.  But a working theory is emerging:

I'm going back and forth between two versions of the host software (trying to be careful and note which one I'm using), the one directly from the master branch using git clone, and one I downloaded manually (2017.02.1).  Both appear nearly identical in the hackrf_sweep source code, including small differences in the do...while loop at the end when the device is streaming.

With the version directly from the repo, the "loop" we get stuck on reports "0 total sweeps..."
With the 2017.02.1 version, the "loop" we get stuck on reports "XX MiB / 1.0 sec = XX MiB/second"

On a related note, when we get stuck in this loop and kill the program, the resulting csv output is always a zero file size, even with the 2017.02.1  version that reports it is streaming something greater than zero MiB/s.

Both do...while loops test byte_count to see if the byte count is zero, and if it is, then exit.  This would seem to work, but somehow it's not and the do...while loop continues.  So the script that is calling hackrf_sweep never "exits" the execution line of hackrf_sweep which is why the next few lines we have to "trap" the error and either exit our loop or invoke the spiflash never gets called.

Relevant code from my infinite (do...while) loop that doesn't exit even though hackrf_sweep starts generating zero size csv files (note: tried a variation on this, except instead of setting LOOP_OP to 0, tried calling spiflash -R, but that fails too).

LOOP_OP=1
while [ $LOOP_OP = 1 ]; do
   hackrf_sweep -f $FREQ_MIN:$FREQ_MAX -w $FREQ_STEP > out.csv
   csv_size=$(wc -c out.csv | cut -f1 -d' ')
   if [ $csv_size = 0]; then
      LOOP_OP=0
   else
      sweep_transpose.py $FREQ_MIN $FREQ_MAX 
   fi
done

My thought on a tweak to hackrf_sweep.c (purely for my unique needs, unlikely to be a viable patch for anyone else) would be to simply not do the (if...then) test in the (do...while) loop, and just set do_exit to 1 at the bottom of the (do...while) loop so that we never do more than one pass through the loop.  (and then remake the whole thing...)

Would that do the trick, or am I just jumping from one problem to another new problem?

closed 2021-02-28 by straithe