greatscottgadgets/hackrf #368: Looping in one-shot mode causes hackrf_sweep to fail

covertcodes, 2017-04-07:
### Steps to reproduce
1. while true; do /usr/local/bin/hackrf_sweep -f100:140 -w 5000 -l 32 -g 8 -1; sleep 5; done

### Expected behaviour
Power levels for each frequency in the spectrum interval given on the command line

### Actual behaviour
Runs fine for awhile, then:
13.4 MiB / 1.000 sec = 13.4 MiB/second
13.4 MiB / 1.000 sec = 13.4 MiB/second
13.4 MiB / 1.000 sec = 13.4 MiB/second
13.4 MiB / 1.000 sec = 13.4 MiB/second
13.4 MiB / 1.000 sec = 13.4 MiB/second
13.4 MiB / 1.000 sec = 13.4 MiB/second
.........
May resume normal operation after awhile.

### Version information
**Operating system**: CentOS 7

**hackrf_info output:**
Found HackRF board 0:
USB descriptor string: 0000000000000000a06063c82365215f
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1
Part ID Number: 0xa000cb3c 0x005d4f4b
Serial Number: 0x00000000 0x00000000 0xa06063c8 0x2365215f

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

dominicgs, 2017-04-07:
Which version of host code are you running?  The hackrf_info output looks like it doesn't match the 2017.02 release (but I can see that your firmware does).

covertcodes, 2017-04-09:
hackrf_info gives this: hackrf_info version: 2017.02.1
libhackrf version: 2017.02.1 (0.5)

hackrf_sweep itself doesn't seem to have a version option.

miek, 2020-01-27:
hackrf_sweep expects to find a "sweep header" at an `n * BYTES_PER_BLOCK` (n*0x4000) offset into a USB transfer: https://github.com/mossmann/hackrf/blob/master/host/hackrf-tools/src/hackrf_sweep.c#L229

If I dump out the received data to a file, I find sweep headers at 0x3e00, 0x7e00, etc. so it seems that a previous invocation is leaving things in a state where the headers are offset from where we expect them, so hackrf_sweep never finds it. I don't really know enough about the USB stack to see what's going on here - can it have part of a USB transfer queued up from a previous run?

mossmann, 2020-01-27:
Yes.  I think there may be two firmware bugs here:

There is a bug that results in the transfer of one or two USB buffers before they have been filled at the start of an RX operation.  The buffers typically contain data from a previous RX or TX operation.  I believe this also happens on TX.  I've observed this in normal RX mode, but it wouldn't surprise me that it would happen in sweep mode.

Being offset from the start of a buffer by 0x3e00 sounds like an additional bug that could be fixed by resetting things at the start of a sweep operation.

miek, 2020-01-28:
I tried making sure that the bulk_buffer_offset & phase get reset, but that didn't help for this issue.

Looking a bit closer, the sweep firmware actually makes sure that the header goes at the start of a bulk_buffer (regardless of where the offset might've been left previously): https://github.com/mossmann/hackrf/blob/master/firmware/hackrf_usb/usb_api_sweep.c#L98

mossmann, 2020-01-30:
Has this behavior changed at all since #692 perhaps?

miek, 2020-02-04:
I think I've narrowed this down a bit further. As a test, I added a count that increments after each sweep_mode() invocation, and inserted the count into the sweep header and also right at the end of the transfer.

When I trigger the bug and see a header starting at 0x3e00, I see the same counter value in that header and at the end of the transfer before it - meaning that all of that data came from the current invocation of sweep_mode() (not left-over data from a previous run).
This suggests that what is actually happening is that the first 512-byte packet of the first transfer is getting dropped. I did some searching around and I'm wondering if it's a data toggle issue?

If I insert a USB bus reset into the loop I can't trigger the bug anymore:
`while true; do hackrf_sweep -f100:140 -w 5000 -l 32 -g 8 -1 > /dev/null; ./usbreset /dev/bus/usb/001/030; sleep 1; done` (from https://askubuntu.com/a/661)

> Has this behavior changed at all since #692 perhaps?

Sort of, but it's hard to narrow down: I'm actually now finding a lot of intermittent trouble with hackrf_sweep (where it'll run slowly, run then error out, or not run at all) in the months before that was merged. #692 itself seems to (accidentally) resolve a lot of that.

miek, 2020-02-04:
Yep! Seems to be a data-toggle problem. If I apply this hack that disables the data-toggle reset on endpoint enable, I'm unable to trigger the bug: https://github.com/miek/hackrf/commit/930bb564b8029c4fa16e55cd751fb463e4feb2a2

mossmann, 2020-02-04:
nice find, miek

mossmann, 2020-02-05:
I've started tracking the other bug I mentioned above as #703.

PR #702 (2020-02-11): Don't re-init bulk endpoints on every set_transceiver_mode call
Fixes #368
closed 2020-02-11 by mossmann