greatscottgadgets/hackrf #370: Feature request: pause between sweeps in hackrf_sweep

covertcodes, 2017-04-09:
### Steps to reproduce
The requirement comes up because of [Looping in one-shot mode causes hackrf_sweep to fail #368](https://github.com/mossmann/hackrf/issues/368) .  Single-shot is not reliable.  A pause is required because some software simply can't handle such large amounts of data per time period (and may not require it).

In particular, my own software is on-hold for HackRF until this is resolved (it works for RTL-SDR only).  The amount of data sent by sequential sweeps is just too much to process for my purposes.

### Expected behaviour
Tunable pause between sweeps, based on a command-line parameter.

### Actual behaviour
There is no pause available.  One-shot seems broken (see above).

### Version information
**Operating system**: All

**hackrf_info output:**
N/a

### Output
N/a

dominicgs, 2017-04-28:
I think this problem has two solutions:
1) We fix the underlying bug that causes looping one shot mode to fail
2) Implementing interval averaging, which will use more data to produce each line of output, and therefore will produce output at a lower rate

Until we are able to resolve those, I recommend working around the problem by simply throwing away some of the data that hackrf_sweep produces.

covertcodes, 2017-05-22:
Agree; FYI I'm throwing away data right now using the following patch to hackrf_sweep.c:

```
@@ -195,6 +195,12 @@
 	return log2f(magsq) * 10.0f / log2(10.0f);
 }
 
+
+// added
+#define SKIP_S  3
+time_t startskip = 0;
+//
+
 int rx_callback(hackrf_transfer* transfer) {
 	int8_t* buf;
 	uint8_t* ubuf;
@@ -207,6 +213,19 @@
 		return -1;
 	}
 
+        // added
+        if(startskip == 0) {
+            startskip = time(NULL);
+        } else {
+            time_now = time(NULL);
+            if(difftime(time_now, startskip) >= SKIP_S) {
+                startskip = time(NULL);
+            } else {
+                return 0;
+            }
+        }
+        //
+
 	byte_count += transfer->valid_length;
 	buf = (int8_t*) transfer->buffer;
 	for(j=0; j<BLOCKS_PER_TRANSFER; j++) {
@@ -633,11 +652,13 @@
 
 		time_start = time_now;
 
-		if (byte_count_now == 0) {
-			exit_code = EXIT_FAILURE;
-			fprintf(stderr, "\nCouldn't transfer any bytes for one second.\n");
-			break;
-		}
+        // gammarf
+		//if (byte_count_now == 0) {
+	    //    exit_code = EXIT_FAILURE;
+	    //    fprintf(stderr, "\nCouldn't transfer any bytes for one second.\n");
+	    // 	  break;
+		//}
+        //
 	}
```
 
 	result = hackrf_is_streaming(device);

straithe, 2021-03-24:
Closing this ticket as the account that opened the issue has been deleted.

closed 2021-03-24 by straithe