greatscottgadgets/hackrf #372: tx rf gain defaults on and if gain only "pulses" to commanded setting.

kf4mot, 2017-04-18:
### Steps to reproduce
1.In gnuradio set rf gain to 0 to turn tx amp off.
2.Start flow graph and see the power out is at -22 dBm on the power meter.
3.Cycle rf gain to 14 and then back to 0 and the power out goes down to -32 dBm where it should be at startup with rf gain at 0.

### Expected behaviour
When the rf gain is at 0 the tx rf amp should not be active.
if gain seems to be stuck at 0 but will pulse up to the setting I enter briefly then go back down to the 0 setting as indicated with a rf power meter.

### Actual behaviour
The rf amp is on until you cycle it on then off again.

### Version information
**Operating system**: 
Debian 8 64 bit.

**hackrf_info output:**
gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
built-in sink types: uhd hackrf bladerf 
Using HackRF One with firmware 2017.02.1 
Using Volk machine: sse4_a_64_orc
gr::log :INFO: audio source - Audio source arch: alsa

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.
GNU Radio Companion 3.7.5

### Output
```
Insert any commandline or build output here
```

kf4mot, 2017-04-19:
Changing from WX gui to QT seems to have fixed the IF gain issue.
The RF gain issue remains.

dominicgs, 2017-11-17:
@kf4mot Is this still a problem?  Could you share your flowgraph with us?

kf4mot, 2017-11-17:
This doesn't seem to be an problem right now, but I'm not sure what fixed it. I'll close this issue.

closed 2017-11-17 by kf4mot

dominicgs, 2017-11-17:
I think it was almost certainly caused by external software which may have been updated since.  I have failed to recreate this bug with `hackrf_transfer`.

Feel free to reopen this if you see it happen again.