greatscottgadgets/hackrf #373: hackrf unable to boot into DFU

theHamClub, 2017-04-18:
I flashed the board as instructed in the wiki and the board worked . It shows up as openmoko which again is perfectly fine however I can't put the board into DFU . In my attempt to correct it I accidentally flashed it with blinky using spiflash while the board previously had hackrf_usb firmware using dfu and now the board isn't recognized either . The board works since I see the led flashing however it doesn't respond to DFU button hold neither does it show up in lsusb,dmesg or dfu-util -l which makes it impossible for me to change the firmware.

miek, 2017-04-18:
When you try to put it into DFU mode, are you holding the DFU button while plugging in the HackRF?

theHamClub, 2017-04-18:
yes , I tried both , holding dfu while plugging in the usb and hodling dfu while pressing reset ,none helped

theHamClub, 2017-04-18:
Fixed it , it was a faulty switch .Should anyone face a similar issue , check the switches too .

closed 2017-04-18 by theHamClub