greatscottgadgets/hackrf #375: Help API

pachoduran, 2017-04-26:
Hi, i am development in Qt creator over Raspberry PI3 with me HackRF ONE, i can connect to HackRf sucess, but i can`t receive data IQ, your can helpme  with info over who commands use for receive data IQ.

Thanks

dominicgs, 2017-04-28:
Could you share with us the calls to libhackrf that you are currently using?

I recommend looking at `hackrf_transfer.c` here: https://github.com/mossmann/hackrf/blob/master/host/hackrf-tools/src/hackrf_transfer.c#L975

pachoduran, 2017-05-09:
Hi, thank you very much for the help you have given me, I was reviewing the document that I have provided and I was able to make good progress, but I have some doubts.
I assume that the rx_callback function is where I receive the information in order to obtain the I / Q data. They could help me as I can record the I / Q data.
Thank you very much.

dominicgs, 2017-05-10:
Correct, the callback function is defined earlier in the same file: https://github.com/mossmann/hackrf/blob/master/host/hackrf-tools/src/hackrf_transfer.c#L363

At that point you have a buffer that contains alternating I and Q samples.

pachoduran, 2017-05-27:
Hi dominicgs, thanks for all the help I have been given.

What I have to do is read 1 time and get the vector I, and a second reading and get the vector Q?

Should it be immediate or is there some delay between the two readings?

Thank you very much

dominicgs, 2017-05-27:
The single buffer contains both I and Q values, they are arranged as `I1,Q1,I2,Q2,I3,Q3,` etc

pachoduran, 2017-06-22:
Hi, thank you very much for your help, I have been allowed to advance a lot in development. I have another concern, when I receive the vector IQ gives me about 200,000 records, is it possible to reduce that amount enough because for the analysis I am doing are too many? 

bye

dominicgs, 2017-06-26:
You can reduce the sample rate to reduce the amount of data.

closed 2017-11-17 by dominicgs