greatscottgadgets/hackrf #377: Undefined reference to - hackrf.h

Cyna, 2017-05-12:
### Test case
#include <iostream>
#include <libhackrf/hackrf.h>

using namespace std;

int main(int argc, char *argv[])
{
    int result = HACKRF_SUCCESS;
    uint8_t board_id = BOARD_ID_INVALID;
    char version[255 + 1];
    uint16_t usb_version;
    read_partid_serialno_t read_partid_serialno;
    uint8_t operacakes[8];
    hackrf_device_list_t *list;
    hackrf_device* device;
    int i, j;

    result = hackrf_init();

    return 0;
}

### Steps to reproduce
1. g++ -lhackrf  main.cpp && ./a.out

### Expected behaviour
Code compiled successfully.

### Actual behaviour
/tmp/ccYle1Pm.o: In function `main':
main.cpp:(.text+0x39): undefined reference to `hackrf_init'
collect2: error: ld returned 1 exit status

### Version information
**Operating system**:
Ubuntu 16.04.2 LTS

**hackrf_info output:**
hackrf_info version: git-9bbbbbf
libhackrf version: git-9bbbbbf (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000a06063c82520335f
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2 (API:1.00)
Part ID Number: 0xa000cb3c 0x00694756

### Additional Info
- hackrf_transfer and other commands work well
- /usr/local/include/libhackrf contains hackrf.h
- /usr/local/lib contains: libhackrf.a; libhackrf.so; libhackrf.so.0; libhackrf.so.0.5.0

closed 2017-05-15 by Cyna

dominicgs, 2017-05-22:
@Cyna Did you solve this problem?  I'd be interested to know what was going wrong.

Cyna, 2017-05-25:
Hello,

yes, problem is solved. It was my mistake. If I just put "-lhackrf" after the file name it compiles without any problems.