greatscottgadgets/hackrf #378: How can i calibrate HackRF one for frequency correction  in DVB-T receiving?

stackprogramer, 2017-05-18:
### Steps to reproduce
1.I had a HackRF one TX DVB-T,in receive it sucks .....
2.with rtl-sdr and vlc it can received DVB-T signal........but with hackrf one sucks
### Expected behaviour
after some research i guess that oscillator hackrf one is not accurate .it can should be calibrate?
how we can calibrate hackrf one for different modulation....................
### Actual behaviour
in real world i guess it can not calibrated in frequency.........
### Version information
**Operating system**:
debian os , gnuradio 3.7.12,linux
why hackrf one not can received ? it has weakness in receiving? weak receiving can be related to ppm ?
please help, any idea???
i will soon share result.
best regards stackprogramer

barto-, 2017-05-24:
You can take a look at [kalibrate-hackrf](https://github.com/scateu/kalibrate-hackrf)

>Kalibrate, or kal, can scan for GSM base stations in a given frequency band and can use those GSM base >stations to calculate the local oscillator frequency offset.

stackprogramer, 2017-05-24:
@barto thanks it will try but we need a kalibrate for dvb-t and atsc.

straithe, 2021-01-28:
Do you still need help with this issue?

straithe, 2021-02-28:
I'm going to close this as there hasn't been a response in a while, but please re-open this issue or open a new one if you still need assistance.

closed 2021-02-28 by straithe

larottao, 2024-05-20:
Hello. I'm using Kal in an attempt to fix exactly this same issue faced by the OP. Key Fobs are seen on different frequencies, and RX TPMS function does not work at all.

Followed Kal instructions

C:\HackRf\kal>one.exe -s GSM850 -g 40 -l 60
meh: Scanning for GSM-850 base stations.
GSM-850:
        chan:  133 (870.2MHz - 34.602kHz)       power: 5526366.97
        chan:  220 (887.6MHz + 27.074kHz)       power: 6232193.72
        chan:  221 (887.8MHz + 37.419kHz)       power: 6177713.66
        chan:  222 (888.0MHz - 13.931kHz)       power: 6549149.89
        chan:  223 (888.2MHz - 31.848kHz)       power: 6391992.48
        chan:  229 (889.4MHz + 19.787kHz)       power: 6663279.50
        chan:  230 (889.6MHz + 19.585kHz)       power: 6427940.00
        chan:  231 (889.8MHz - 30.238kHz)       power: 6751701.10

C:\HackRf\kal>one.exe -c 133
meh: Calculating clock frequency offset.
Using GSM-850 channel 133 (870.2MHz)
average         [min, max]      (range, stddev)
- 20.295kHz             [-36662, 6386]  (43049, 11513.176758)
overruns: 0
not found: 2176
average absolute error: 23.322 ppm

C:\HackRf\kal>one.exe -c 220
meh: Calculating clock frequency offset.
Using GSM-850 channel 220 (887.6MHz)
average         [min, max]      (range, stddev)
+ 31.634kHz             [25639, 37408]  (11768, 3628.375244)
overruns: 0
not found: 359
average absolute error: -35.640 ppm

However the deviation shown on the documentation example is only 1 Hz, and mine is huge. On the other side, I'm not sure how to enter this offset on the Hackrf offset menu. Any guidance will be very appreciated.